diablodale/dp.kinect2 #6: Registration key file or name not working

ChrisOnCode, 2015-01-14:
Setup
Desktop PC Intel i7-2600 CPU 8 GB RAM, 120GB SSD drive
Windows 8.1 x64 with all current updates
Kinect for Windows SDK v2.0_1409
Max 7.0.1 x64 (currently in trial)

Steps:
1. Open dpkinect2 patch (v.0.34.0.0)
2. Max Console prompts "dp.kinect2 is not registered or has expired. Register your key or visit the software section of http://hidale.com for a new registration key."
3. Register dp.kinect2 by entering your registration name in the top text box, clicking the register button, and choosing the registration key file

Result:
Max Console again prompts "dp.kinect2 is not registered or has expired. Register your key or visit the software section of http://hidale.com for a new registration key."

Expected:
Some message that confirms the registration

Additional information:
Also tested on Max 6.1, same behavior

diablodale, 2015-01-14:
please read the stick issue to provide more detail. There is not enough information in your above entry.
https://github.com/diablodale/dp.kinect2/issues/1

ChrisOnCode, 2015-01-15:
Hi Dale, thanks for getting back to me. 
I've read the stick issue and added some steps.

Please tell me what other information you need.

diablodale, 2015-01-17:
It could be there is a bug in Max itself and not in dp.kinect2. I have found an issue with how Max objects (zl) and (tosymbol) processes message lists with commas in them.

Are you Chris from Lübeck?
If yes...then your registration name has a comma in it. And the zl/tosymbol objects are not handling it as expected.

A quick workaround is to send the registration message yourself. This is documented in the wiki.
Here is a button you can use. Paste the below back into your patch and connect it directly to dp.kinect2.

----------begin_max5_patcher----------
290.3ocSPEraBCCC8bQh+AqbtqhxPnxtsiHscZZmFSSgVWVf1jpDWTYn8msa
6GaINq.WRz64m8y9cd5jDwVy.5DvCvaPRxYOSByEXRFIRDsxgxFoiEJZQmSt
CEo+WjvAhKXwcJGgVXiXs1QYPcF7beCoZwJkrAgdcErV6UHOPpiH7xIu7VLE
dU6gVmhNAlZ3oe+YKVdXi3hE59VSO0fDuA4izcRp7Skd2GVrjhGwxh4YyRg7
Ug2E42yfhrYv6iMEGDcpCicHDWqUazjS8EWIOLnq9qzi1O+V0ZYKqV7n0ehW
VXUESZ1t+t7BAS98zIge+WzOgrqiOZilSaNq8I8diMfWlFwJcDGOZeFeTM1x
pHkz5CAxm.813pLrbQvR1P+yejl04MG
-----------end_max5_patcher-----------

If you are not Chris from Lübeck and your registration name does *not* have a comma in it then please read the following.

I am not able to test in Max trial's. Their copy protection disallows re-trialing. I doubt there is a problem with that topic, yet I am not able to test it.

Please provide additional details on each step. Pretend I have no knowledge of anything. I need to know on what to click and what to type.

* Do you start Max first? How do you do that?
* Do you doubleclick the patch file in finder? Do you choose fille/open?
* What is the registration name you type in step 4? (If you don't want to provide that here in the issue, please send it private to me in email dale@hidale.com and please reference this issue)
* What is the name of the registration file you select in step 6? (Please do *not* attach that file here)

ChrisOnCode, 2015-01-19:
Yes that is me. Thank you for providing a workaround! :)

As I haven't checked github over the weekend, I've already registered a trial (with a short registration name, no special chars) to start testing. I'll try your workaround soon.

diablodale, 2015-02-07:
The demo patch was updated and the workaround is available available for end-users to sent the registration message themselves.

closed 2015-02-07 by diablodale

ChrisOnCode, 2015-02-23:
Thanks again for your support! Works like a charm : )