diablodale/dp.kinect2 #7: Camera stops working after a while

dietervandoren, 2015-02-07:
Windows 8.1 Pro 64bit updates applied
Max 701 32bit registered
Kinect SDK v2.0_1409 (Kinect SDK v1.8 also installed)

Gigabyte GA-Z87X-OC
i5-4690K (no overclock applied)
Corsair 8GB 1600MHz
GTX670 graphics
MOTU Ultralite mk3 hybrid (connected by firewire)
Dawicontrol DC-FW800 PCIe firewire card
Sitecom N300 USB wifi dongle
Sparkfun (FTDI) USB Xbee receiver

I 've had 3 times today that I left my perfor;ance patches running while doing other stuff (not on that computer) and when I come back the kinect 2 has stopped working. The lights are off and dp.kinect2 doesn't output anymore. No error messages in the Max window.

When I send 'close' and 'open' to dp.kinect2 to re-establish connection I get in the Max window:

kinect1: close 1
dp.kinect2: Operation aborted. hr=0x80004004
kinect1: open 0

It won't reconnect before closing the patch, which crashes Max entirely, and re-opening.

The kinect patch is copy-compressed below. It is part of a larger setup comprised of dozens of other patches and dependencies. Can't send all of that. I'm currently testing whether the issue occurs when only the kinect patch runs.

<pre><code>
----------begin_max5_patcher----------
6570.3oc6cszaiqjcdcO.y+ABmF2EC7UtdWEmEA8LSxhY0M.CR1by.CZIZad
aYREIp9wLH42dpGjRjxTEOxsJRJa2Fsj36S8wScdUm5T+ye+u6CWcWw2R2bU
zeL5Wi9vG9m587A69L64C063CW8Tx2luLYi8DuJO8qE28aWcc0wJS+Voc+qh
9ZV9tcmu8or7kok1qAWu26KxKySdJ0dA+o0YIKadAEaKquBT8tyVXOW8S7mE
7cm7pjx4Olk+vsqSmW5H+XJZF55HpPMiq+J1rAgLCE82a9vuOYdZq6uYmax9
G1ch0md892T98kNxr8CMccE1TCN5aQ1xzujtdSVQdyF6GtJY0pF6+CMuHCl9
aE16k7586KK2sOz98sN8KYO6Vqu2q0.Pot0ucsiL+lRbUi6Twhz04ayrWU0d
+e+8+tcj1dbiYZzQXgEvvbJ17shiafbZVhGVVL+yoKZQaWUrJMOKe05zMo4k
IkUD49iuH89jsKKu8HXb6S3v2LsOZ2LMe3pGVmsnH2PHsgdy9qej+ptY4Zbs
ZS1SIOYUWWtlcSCPG6nazM1satKYs4MycN1DxtiVVTrr8w1ekKSuur53qxxy
ODPKKV44nqyd3QeW8cE5i9j2au8Patcat6v2p6TWd6ljub.xWlrbYU27CdBe
KIO6ojxzxL26CBZ+QSySzM3G2Lecwxksa2tC8ktNzBM+87zulsn7Q6CqE6g9
BxVUyXc09W6KxdHcS4A6rL4gMGrqm2KVuus2U0S91xzmVsT2ZN3LZITrUu1l
BGaefCDRZk8su+3gBDQGbjFR9vMOTCoe3V2sND.hkVlb6mUxBQMY3s701mS4
2WU0w3peKq7V86y0Y5VUqyrCjyr64EO8jtyt6.61uVxR8uMBYdwnllZlsRqE
YQwWON10BfZJ7oAVXa91OHXmfMA+PrnMrSNBryP8g6BxdbGiDrYLIQJUZgnH
Ru3+UWq+eze+HHYCoUKyxOJOoknLmvQv3MEaWOu9IVwKEc.ko6qUlkuSH9ut
u0e3Y9X1hEo4OiKdQ1FSebmHmi1dbarWSzUFgOKt0Iu81jRMm3caKcszVJYO
QYAZ8U2krrpe9NkG9ELrW9wNpz9ipuq3rgYjzVcujs8XLTaFPZGV7DS7YwCF
wDFlNtx9okAjPaZwyyX2xzccudGW2968ZMBUpEH5jSebCjn6MPRa7wSULjW8
K2eu4Vds4ieIuwctAL+hgRM+PYQ9Igk3tvRLDrzYo.AU+oGr7tj7GNyM0iYZ
cxBmIGQa9Zllz6FKHmrQ1jNfIULDXhZ+Ll9bir6TBmgsqsc38Xx8KFAKKd3A
8M4GkYQo75pAuAH.hWoM.3s+VH4iVVjr3ozMahPmK+z5B7DL.fmqilj.gCZX
XcNFnsHY9+24pGWmt0h..WNY7TY2d0NzPxpn4IKyt61ujk90P5wO2a2PhrxJ
WqKqRwKBadc6bOwZ7K2hSXJlYkZaPp28s+ce6e229WlWpFMnIOj1xuvchFML
l5NBJTTr.cbmXa4tomNAviOfpO+TcRK4tv6Y6yn528z1G+Xc16Rey4Lt.8i3
BZ7PC3x9.bhfsCwkwu1PbKONSFOjPtn2Xf4Bi8EIS9pj7z1fxWR1G5hrhe1d
B+Z7eGXnwtqX8BmAM3iizniwbS6MrWVaubw9RyJX9lY3xaClNhXdwRmoOUQl
C27i1Wv5hs4KNP4lm3QZLE5Ymq1w7JmpZ+16tGZPH31DxyBU5p0EqJVuyvmY
z3w5MtZPdiKhOg23B23qok399K7wM9108853UAr.bSdAcz63gchQ195KoQVw
h.Vc4Dd8miv3pDpQCPAcz.Hm0AC35y.oKgR5FoaSLZW.k1M5hmXzNdbXYd6L
9QGKxaZskkoZDE2cX2nmbX25ZXm3Regci4bkwIJzEWR.CAvAC4THiZYsrzmA
dnnGSSVD8emiixSm+Y8OHQkEq2Tn+EMZ8sadrX6Rs0K5MYQKatIWezzk2U7U
8uE5CU+aod+OljatmJ8tq9YrYuYqLOHjYuteh068y4ooleSz6t92lm78EEkl
eadrleui1McJmqsPx1.v7.FqUlv6KcsAeb8acLhZ+FIMeQvzgJTzG8k5+4e6
OGYraK5eIjQhlQ8gNRgblzfGMFSCzPgLG3p9djgdtF0hNGjGuiSnByc.Bq1G
7ywn7XZYG5YQCKgad66HB+28v8YKWZu9aqetW8v5jEYM4qZdV1OwG58vyeXO
6ZH0DnPwu9fu7ek0WHIFgiElKhhhUHl6W5cg8bG14FjajQ53LdtONcde1VVr
CXrmmL97ydRBI6oz6fpIisuIem67ctyiwchCJ2Iwu1jVBOsJdem87MI6oe2.
vQjninjmcx7ort3S494ScYAjMoDDmji.Cmy.5F4coqavg1YDAeNa4OnMjckZ
URFjd8tAqm4TPI6ENOHKz7lZQZmia0OQZ6Xs+S7QMYUr+kxiGdWTRy0qHlOI
rCtKuLoNm89KeI5yY4zPlzSRuNHnXN2lbYLG6BHqmrHFInHl.BhI3WTHFNnH
lBBhQmbH1gFYEBKp7m1pXVU54yqLoZxhMMiqw+0e8e6VDCE+S+Glef3+I4Oo
[... 81 lines cut ...]
HgEC+kIcfIMEbRaf8FD1H53HM9.SZB3jFafIMNLu6GitAfRR5mI5avHs9MWf
EORjV+1XwTiDoI5mzjiDo0uUVLwIm1ZBxLrfgMkiVoaMbpZWTjrZWAzjEXCZ
KgMBNpynSWRCngdiAogfSZCrgdPGlT9va1BUAmzFXyV.NRoJ9va1BU.mzFXy
Vfku3B1v2AkBJqMsTFd3oLBLJiL7TFEFkQGdJiAixFgd.bXTFe3oLALJSL7T
lDFkIGdJSAixTCOkECixhGAIs.UBfGCs.PUCLB5Av.UDfGAMAXfpBvift.LP
kA3AVa.gAJLNiPZs3Hs9CiyHTPEbjFfQ8IdjHM.CUlZjHs9CiCUNRjFfAK6j
CiifH2GFGjcDnDDwrXQrhEWsq.FFGX963Z9CMjSltjFnoDCehRXiAhggEmj3
QHii.ERN5vqDAGqlpYsLNVNcIMXwVJlMBjFe5hZrfVo4HzlSVh1aMjI85yJt
fGi6fLBuBfk+3i.gEx7wUzLebEAMebwJfBTGdkPXfUa0QHK8wJfRsFANSXUT
T1IWD14X7rXFAgkWGoPthABAaKK6lsBIKJvwefOBfcPKgWCpVffV3knMknQC
qDMAOLc.HLgsC.kLzc.fUIovhQPAMrxrDeDrdSDz4DaUTIplErs2ZxXXGOP1
uZqMTLZild6szlJ1rUFGzrtACKKIIiAOnkz.DMV1HQZpoKoAH75zQhzhmtjV
+7ZD5jkWazHs9ig8HL0ogJWavSkeLzzRWMUEcfYhoZ+S7HjAlNRq+wAdzzEz
+v.ylrhNvrwRzQ+YEzHL.0.0EfGgAnFnxcLcrLjr+johNZhbAT5LGgvzAbp.
AXNPbXr.vw1Xyhv00F25sBoiN7P0ZjiQqgEnVCkpFgVCMPsFFRNiRorXgtQf
40kkYd0VgrIQBUSRQmgoBJmN3ukvgpIESlIvXjhM3ukPApIIXjQnaTPCZGuU
n33iWf4.N4cN8WZRrbGS2fwBRCYXFwMJITs1XXCxHInoBvjgwDj7QwH31.Ho
bifg4P7NM9TWi3jJ4LkVoOldscTmrK4BUwYu0Vb6WwUcU3AMsHwPboTcxKxL
XraLzHTqdFI2NvZGrE1UfZcqkos2pRLfrpmTfEC.ZIFlLJrg85AlPMMoK93P
W8xKCnTG3mWVn6FiNbqZ9UDdf3W6MtdiPE..Ctpqimd7EzeX1BVK1BWcVRX0
1qbV3zdKg8yp52sfFZFF.gz7GqGq6BuJY0pujtdS081QSW8TxuUr1rs7Z21Y
4tsc24qVm9kr5KA61Ux54OlUlNub6ZaS4puYRbL2UWrHcc91r5g9vRC5O9+Q
oBGqX
-----------end_max5_patcher-----------
</code></pre>

dietervandoren, 2015-02-07:
It doesn't seem to hang when only the kinect patch runs. I'm trying to find which other patch causes it, starting with the ones that receive the 2 jit.net.send's matrices and udpsend's message.

diablodale, 2015-02-08:
OK. When you have a repro, please do send it my way and I will try to reproduce and then isolate it.
You might want to test additional things
1) Run Microsoft Kinect Configuration Verifier at http://www.microsoft.com/en-us/kinectforwindows/purchase/sensor_setup.aspx
2) Run one of the Microsoft supplied Kinect samples and run it on teh same machine for the same time and manner in which you are having the problem.

dietervandoren, 2015-02-09:
It seems to be a Max issue, not dp.kinect2. My Kinect patch exported as stand-alone keeps running steady while the other patches in Max hang. The previeously stated symptoms of dp.kinect2 malfunctioning appears to be consequences of something else hanging Max, not the other way around.

diablodale, 2015-02-09:
Got it. I'm closing this issue for now. If you isolate a repro case in which dp.kinect2 is failing, please re-open this issue or open a new issue for it.

Perhaps as a help... There is a block of dp.kinect2 code when you "open" the Kinect that can return a 0x80004004 error. Such an error is likely due to:
1) a problem in Max which forbids me from successfully allocating memory structures
2) a problem in the Kinect runtime/drivers which forbids me from successfully querying for camera alignment information

closed 2015-02-09 by diablodale