diablodale/dp.kinect2 #8: trouble loading the ddl for face tracking

wadox, 2015-04-11:
the max window says:

dp.kinect2: dp.kinect2 requires Kinect20.Face.x64.dll 2.0.1410.19000 or greater + the matching NuiDatabase folder from that release

I have downloaded the latest versions from your website: dp.kinect2-v0.35
and face tracking models: dp.kinect2-v0.35-face-models

I saved the unzipped files in the same folder as the patcher, I am using Windows 8, Max 6, other functions are working.

diablodale, 2015-04-12:
Please read issue #1. I need far more information in which to assist you.

wadox, 2015-04-13:
I have now updated the required above

diablodale, 2015-04-16:
What step-by-step do you do between "I saved..." and "result". Tell me everything. Pretend I am blind and old and don't know anything.

Meanwhile, please ensure that two files and one subfolder are in the same directory.
dp.kinect2.mxe64
Kinect20.Face.x64.dll
NuiDatabase\

wadox, 2015-04-17:
Thanks dale I forgot to include the object , I have face properties outputting into the max message window now , to get it to work I included @faces 1 and @faceprop 1

Regards
Dr. Wade Marynowsky 
www.marynowsky.net
0432 940 562

diablodale, 2015-04-17:
I'm glad is it working for you.
If you begin using the HD face model features (HD 2d points, 3d vertices, SU, etc.), please be aware that the Microsoft recognition technology is very very slow compared to Kinect v1 and needs controlled slow scans of the left, right, top, bottom of the face before it can construct the 3D model.

I'll have the dp.kinect2 documentation updated/completed at https://github.com/diablodale/dp.kinect2/wiki for the final release in a few weeks.

closed 2015-04-17 by diablodale

diablodale, 2015-04-17:
Updated wiki setup section to add clarity