diablodale/dp.kinect2 #10: Registration Problem

kenbyers, 2015-06-09:
I have a dp.kinect2 professional registration only a few days old, which worked for a short time in Max 7 64 and MAX 7. I have rebooted win 8 on my computer then have tried many times to register dp.kinect2 and I get error reports in Max 7 console show in console screen shot. 
I then followed help, to unregister, which does not open, and to put my registration name in " ", as shown in help. Still no registration with dp.kinect2.

Computer details:
i7 cpu. 24 Gbt ram. NVidia GeForce GTX960 SSC Graphics Card. 
Win 8
Max 7 64 , Max 7

I have included screen shots of dp.kinect2 patch and max console

Could you help please

Thanks,

Ken Byers

[image]
[image]

diablodale, 2015-06-09:
You will not need the two quotes around your name when you use the demo patch distributed with dp.kinect2 to register. The textedit field there (and some hidden objects) do the work to keep your registration as a single string (aka symbol) rather than splitting it apart and making multiple symbols. Given your steps and the screenshot, I suspect this is the issue.

Please try again typing only your registration name without quotes around it in your textedit field.
If this doesn't work for you, please use the demo patch which is distributed with dp.kinect2.

If you continue to encounter problems, please provide me step-by-step instructions. Click-by-click. As if I am completely uneducated with Max or dp.kinect2 so that I can reproduced your steps here and help you resolve the issue.

kenbyers, 2015-06-09:
Hi Dale I did what you said and this did not run for both registered and demo dp.kinect trial.
I went to Win 7 on my second drive where I have your email & posted new  comment to you.
I then went back to Win 8, opened Max 64, and then opened my patch, and clicked on register ( I have done this numerous times) did not run. I then opened dp.kinectv2 demo, then clicked on trial, and it worked. I then closed demo and opened my reg patch, and clicked on register, (without quotation marks, as previously done numerous times before) clicked on the DREG and it success it worked.

This is very odd as I have done the exact same thing including trying dp.kinectv2 demo many times before.

Thanks

Ken Byers

diablodale, 2015-06-10:
Everything works now?

diablodale, 2015-06-27:
closing due to no repro

closed 2015-06-27 by diablodale