diablodale/dp.kinect2 #11: dp.kinect2 registration

kenbyers, 2015-07-01:
Hi Dale

Same problem again!
I am running on Win 8.1 with Kinect v2 all working.

I open Max 7 as I want to run OSC and I cant do that on Max x64. I open my registered copy of dp.kinect2, click on Ken Byers my registered name, then click on my dpreg, then it comes up again, so I click again then click on start, not working. Max console error either not registered or registration expired.
I then click on trial version which I newly downloaded as you suggested, trying both exclamation marks on " Trial", and just Trial, clicked on start, nothing running both times, same errors!

Hope you can help

Ken

ken@interactionalmedia.com

diablodale, 2015-07-02:
From your description, I see no failures. It seems to be doing what it was designed to do. Trials are one-time for 14 days. Multiple trials are not granted.

I will need much more information. Please reference https://github.com/diablodale/dp.kinect2/issues/1
step by step.
Everything as if I am a 90 year old person that has never seen a computer before.
Every install. Every click. Every download and version. For example, it is unclear what Max you are running.
Every file you load or access. For example, what is the filename of the registration dpreg file?
Exact text of the Max window.

In addition, please enable `@verbose 1`. This additional warnings/error information can help you and me to understand why registration is not allowed.

kenbyers, 2015-07-02:
Hi Dale

Computer and OS

x58 Motherboard, x64  Intel i7 CPU 920 @ 2.67GHz 2.67 GHz
Win 8.1 x64 
NVidia GTX 960 SC
Transend PDU USB 3.0 PCI card
Kinect for Windows SDK 2.0
Kinect v2
Max 7 registered
dp.kinectv2  - dp.kinect2-428.dpreg

1. I clicked on register my "Ken Byers" , without inverted commas.
2. clicked on dp.kinect2-428.dpreg
3. clicked on  dp.kinect2-428.dpreg as folder open again after clicked.
4 clicked start 
5 rgb and depthmap did not open

Hope you can help

Ken Byers

Please see errors

[image]
[image]

closed 2015-07-02 by kenbyers

diablodale, 2015-07-02:
I believe your issue is you are trying to use your dp.kinect2 license with an old beta. The license you purchased is good for one computer running v1.x of dp.kinect2. The screenshot above demonstrates you are trying to use it with an old beta v0.35. Please download a current version of dp.kinect2 (the most current is v1.0.1) from https://hidale.com/shop/dp-kinect2/ so that you can use your license.

If you continue to encounter problems using v1.x of dp.kinect2, please repeat the information you provided above using v1.x and we can continue troubleshooting.