diablodale/dp.kinect2 #12: Kinect Development without a Kinect Issue

jamesthian, 2015-07-21:
Set up:
-Windows 8.1 x64 via Bootcamp on Macbook Pro Retina, 500GB SSD, 16GB Ram, Intel chip set
AND Windows 8.1 on z400 Workstation, Intel Xeon, x64 OS and processor, 18 GB RAM
-SDK Browser v2.0 and Kinect Studio v2.0 installed on both.
-Max 6.1.7 installed on the workstation, Max 7.0 installed on the macbook. Both unregistered. 
-dp.kinect2 registered and installed

Steps:
Open the XEF file in Studio.
Go to the PLAY tab
Click the Connect to Service button
Click PLAY

Open Max patch with dp.kinect2 object (the one provided in the download). 
Click toggle/open. 

Result:
Max freezes and goes unresponsive. 

Expected:
the dp.kinect2 object to show the streams from Kinect Studio v2.0, as per Kinect Development with out a Kinect (http://blogs.interknowlogy.com/2015/01/14/kinect-development-without-a-kinect/)

diablodale, 2015-07-21:
What version of dp.kinect2 do you have installed? Please copy/paste the full text describing this from your Max console window.

Can you successfully use dp.kinect2 using an attached sensor (not using Kinect Studio)?

I have tested usage of dp.kinect2 with Kinect Studio. It worked in my test scenarios; however, I did have to write some code to workaround Kinect runtime bug/issues which have long been reported to Microsoft and not fixed. It could be your scenario is another special case that I'll need to handle.

Would you please upload the XEF file to some public location and provide me a link to it? I ask because each XEF file is unique and I want to reproduce your issue so I can isolate any problems.

jamesthian, 2015-07-21:
Thanks for the speedy reply.

I have:
"dp.kinect2 v1.0.1.0, Copyright (c) 2014-2015 Dale Phurrough. This software
comes with ABSOLUTELY NO WARRANTY."

And I can successfully use dp.kinect using an attached sensor.

And here is the link to the XEF file:
https://www.dropbox.com/s/xjhjlbk19zx86nd/20150721_165833_00.xef?dl=0

In theory, would I be able to run both a live sensor and the XEF file
through two dp.kinect2 objects with two unique usbib's?

diablodale, 2015-07-22:
When opening the Kinect with dp.kinect2, there is a multisecond pause/hang while the code looks for the Kinect or Kinect Studio. When using Studio, the pause can be longer than with a physically connected Kinect due to workarounds for Microsoft bugs. During this pause/hang, Max cmay be unresponsive. It should never be more than 11 seconds.

After dp.kinect2 tries to find a physical Kinect or Kinect Studio, it proceeds with the data output requested...or reports a failure.

The XEF file you provided has only the body joints, depth, and IR. It does not have color data recorded. This is relevant. Why? Because the demo patch wants color data to be output. Since there is no color data available in the recorded stream, you will get no data of any visual type output because the requested visual data is sync'd with each other. If one is not available, then nothing is output. Try enabling @verbose=1 and you will likely see a repeating error declaring no data available.

I recommend that you disable the color data output on the demo patch by removing @colormap=3. Or...record the color (and other) data so that it matches your output needs/settings on dp.kinect2.

To reproduce and isolate what I describe above, I used Max 6.1.9 64-bit on Windows 8.1 x64. Then with the v1.0.1.0 dp.kinect2 64-bit, the demo patch, and your XEF file.

Does this explain your experience and provide a resolution?
I did encounter a few second hang in Max while dp.kinect2 searched for Kinect Studio

jamesthian, 2015-07-22:
Yes, that explained everything. Thanks so much.

diablodale, 2015-07-22:
Glad to hear. I'm closing this issue.

closed 2015-07-22 by diablodale

diablodale, 2015-07-23:
I missed one of your questions. You asked:
> In theory, would I be able to run both a live sensor and the XEF file
> through two dp.kinect2 objects with two unique usbib's?

Microsoft does not support multiple Kinect v2 on a single PC. It originally was due to one sensor saturating a gen2 PCIe bus and making it technically impossible to handle the bandwidth of a 2nd Kinectv2 sensor. Due to this, they removed all functionality from the runtime/APIs for multiple sensors. Therefore, dp.kinect2 does not support multiple sensors on the same computer.

You do bring up an interesting scenario (one real sensor + one playback). However, it is impossible due to Microsoft core limitations. I do not expect Microsoft to change this in any reasonable timeframe.

jamesthian, 2015-07-23:
I figured as much after some trial-and-error (and after some digging
around).

Found a workaround, though: dp.kinect2 gives you everything you need to
record (point clouds, depth maps, et. al.) for playback with a live sensor.