diablodale/dp.kinect2 #13: orientformat 2 matrix hierarchical (not supported)

jeremybailey, 2015-10-04:
Confused why I can't seem to select orientformat 2 matrix hierarchical, it says (not supported)

"matrix hierarchical (not supported)"

would really love my data in this format instead of quats.
is this a known issue?

diablodale, 2015-10-04:
It is...not supported.  ;-)
Microsoft dropped that feature on the Kinectv2 SDK.
Because of the very low need, I did not implement it by hand myself.
It is possible for you to calculate these yourself using the hierarchy of joints + the quaternion hierarchical values dp.kinect2 provides. And max has objects, like jit.anim.node, which can accept the quat hierarchical rotations...they can even output matrices if you want them.

diablodale, 2016-03-02:
Hello. I added a full suite of joint orientation data to dp.kinect2. You can test it with the v1.1 beta from the usual download location http://hidale.com/shop/dp-kinect2/

closed 2016-03-02 by diablodale