diablodale/dp.kinect2 #15: dp.kinect2 registration problem

Guito1986, 2016-01-14:
Hardware:
Laptop
ASUS ultrabook s46c
Intel icore i7 1.9GHz
nividia 635m 2gb
6G RAM
1T HDD with 24G ssd

Software:
Kinect SDK 2.0
Windows 10 up to date
dp.kinect2 v1.0.1
Max 7.1 64bits trial for now (but it works before on another PC)

Steps:
I download the dp.kinect2 unzip it
Take everything and put it in file on my desktop
Open dp.kinect2 
erase trial in name put decod.ca
click on green button register
search for the file coming with my purchase order 1386
And Max COnsole say: not registered in a red line

Guillaume

diablodale, 2016-01-14:
Do you have the verbose and console output I requested?

Guito1986, 2016-01-14:
Console saydp.kinect 2 is not registered or expired

Guito1986, 2016-01-14:
dp.kinect2: dp.kinect2 is not registered or has expired. Register your key
or visit the software section of http://hidale.com for a new registration
key.

Can I have another registration key just to try?

Guillaume

Guito1986, 2016-01-14:
The point is this I need today we are already late for our deadline.

GM

Guito1986, 2016-01-14:
dp.kinect2: dp.kinect2 requires Kinect20.Face.x86.dll 2.0.1410.19000 or
greater + the matching NuiDatabase folder from that release

I tought it was for the face tracking

diablodale, 2016-01-15:
Hello. I do need the verbose output from the console as I requested in our email. I'll be more explicit:
1. quit all applications in windows
2. find the folder which contains dp.kinect2.maxpat, dp.kinect2.mxe, and dp.kinect2.mxe64
3. double-click on dp.kinect2.maxpat
4. at the top right of the patch, find the attrui called [verbose]
5. enable verbose output by checking that checkbox
6. at the top center of the patch, remove all text from the registration name box.
7. type in your registration name. Do not put quotes around it. Ensure that there are no extra spaces before it, or after it. The registration name associated with this order has a punctuation character. Please be sure there is no spaces around the punctuation.
8. Press the register button
9. Select the xxxx.dpreg keyfile associated with your registration name., then click Open. You must use matching registration keyfile and name. Sometimes you will see this dialog box twice (this is a max bug). If you see it twice, please be sure to select the dpreg keyfile each time.
10. Using the menu bar, choose window/max console
11. Click on any line in the console
12. Press ctrl+a
13. Press ctrl+c. The contents of the Max console are now in the windows clipboard
14. Switch to github, open this issue, click in the area to type, press ctrl+v
You will now have all the diagnostics data pasted into this issue.

You have downloaded the keys twice on both orders. I have checked the keys on the server with your registration name, they work perfectly on my test machines. I do not believe the key download was corrupted twice. However, I have enabled a 3rd download of the key files for you as you requested. You can use the download links provided in the email sent to the purchaser.

Guito1986, 2016-01-15:
dp.kinect2 v1.0.1.0, Copyright (c) 2014-2015 Dale Phurrough. This software comes with ABSOLUTELY NO WARRANTY.
-All rights reserved except for those: a) documented in the dp.kinect terms at http://hidale.com,
-and b) granted by purchase or agreement to any entity listed as registered below.
-BY USING THIS SOFTWARE, YOU ACCEPT THESE TERMS. IF YOU DO NOT ACCEPT THEM, DO NOT USE THIS SOFTWARE.
dp.kinect2 is not registered or has expired. Register your key or visit the software section of http://hidale.com for a new registration key.
dp.kinect2: dp.kinect2 requires Kinect20.Face.x86.dll 2.0.1410.19000 or greater + the matching NuiDatabase folder from that release
dp.kinect2: dp.kinect2 is not registered or has expired. Register your key or visit the software section of http://hidale.com for a new registration key.
dp.kinect2: RG: 33816576 RV: 0
dp.kinect2: dp.kinect2 is not registered or has expired. Register your key or visit the software section of http://hidale.com for a new registration key.
dp.kinect2: RG: 33816576 RV: 0

diablodale, 2016-01-15:
Thank you. This suggests there have been some unexpected changes in the user/machine account or computer configuration. 

On the problem machine:
1. Did you ever have an earlier version dp.kinect2 (betas?) installed?
2. Have you added or removed the machine from a domain?
3. Have you migrated user settings from one account to another?
4. Any login account changes?
5. Any lockdown, protected, or other user restrictions on the logged in user account in which you are registering dp.kinect2?
6. Any reconfigured or migrated domains using toolsets? Actions by domain administrators or tools that are transparent to users/customers may cause registration to fail.
7. Any hacking or circumventing of dp.kinect2 settings, configuration, or registration?

Guito1986, 2016-01-15:
I format that PC and migrated m'y Microsoft account

diablodale, 2016-01-15:
What do you mean migrated your Microsoft account? Please be specific and technical.

I am trying to understand how some dp.kinect2 registration data got on to the user account of the problem PC. Typical Windows 10 settings sync will not copy this registration data. It had to be something else.

Guito1986, 2016-01-15:
I mean that I reset this PC that was the laptop of my boss and when it was
done I enter my Microsoft username and password and synced my Microsoft
account.

diablodale, 2016-01-15:
At this point, I can provide you two options:
1) I can send a method to reset the registration on this PC. This will probably resolve the immediate issue on that specific PC. We will loose the opportunity to further debug this issue.
2) We can continue to debug this issue. The diagnostic codes and that which you wrote above suggest that registration information from another PC was copied to the problem PC. This is suspicious and won't work by design. Registration must be done using dp.kinect2 using the documented methods.

Which of the above options do you want? Or do you have a 3rd option to propose?

Guito1986, 2016-01-15:
The first one look like the best option. Does another account with
administator will solve this?

diablodale, 2016-01-15:
If you log into the problem PC with another account and then register dp.kinect2, I suspect it will register.
Do you want to try that first?
Or do you want me to send the reset method of option 1?

Guito1986, 2016-01-15:
I'll try that first but in case just send me the other option. I'll get
back to you. Thanks a lot. Sorry for yesterday my boss was up my a** and I
was litterally panicking because of the stress.

diablodale, 2016-01-15:
I have sent a special "reset" license to the same email address that made the purchase.
Within the next 48 hours, on the PC with the issue...
1. open the demo patch
2. Use the registration name that is also sent to the same email address.
3. Click register and select this special "reset" license in the next 48 hours.

You should see in the Max console window that your install has been reset.

Now you should be able to use your regular registration key on this PC. Please follow the normal registration steps as at https://github.com/diablodale/dp.kinect2/wiki#register-dpkinect2

Guito1986, 2016-01-15:
It work thank you

closed 2016-01-19 by diablodale