diablodale/dp.kinect2 #17: Error 126 _ runtime reinstall not working

singularmoments, 2016-03-07:
Error 126 _ runtime reinstall not working

Setup
Shenker (Clevo) W705 laptop with 16 GB RAM, 2 x 500GB SSD drive, Intel i7_4790S cpu
Windows 10 x64 with all current updates

Installed Kinect SDK v2.0_1409
Speech and language pack US ES FR
Installed Max 7.1 x64 and it is registered

Steps

Quit max7, I removed only the 1.0.1 externals, and put the 1.1b externals and restart Max 7.1
all the others kinect files in Max 7 folder stay the same
in 32 and 64 both give:

Error 126 loading external dp.kinect2

The kinect is working fine in :
Configuration verifier
Kinect Studio v2.0
and modules of SDK Browser (skel speech etc..)

tried to reenter the Registration Name
no luck

uninstall and reinstall kinect runtime
no luck

uninstall and reinstall kinect drivers
no luck

reinstall kinect SDK
no luck

If i re-place by the old dp.kinect2 v1.0 externals, Max7 works fine.

singularmoments, 2016-03-07:
I re-put the old dp.kinect2 v1.0 externals

and Now, dp.kinect2 in Max 7.1 32b and 64b gives a console :

dp.kinect2: dp.kinect2 requires Kinect20.Face.x86.dll 2.0.1410.19000 or greater
 + the matching NuiDatabase folder from that release

before it was not putting this error.
but the dp.kinect2 works fine with all maps, skel etc ...

I unchanged the dp.kinect2 folder
It still contains Kinect20.Face.x86.dll 2.0.1410.19000
and the NuiDatabase folder , from your install

What should i do?

singularmoments, 2016-03-07:
Kinect20.Face.x86.dll does not seem to be the issue.
For info, the externals as to be in the same folder than Kinect20.Face.x86.dll, to find it.

Still no luck with dp.kinect2 v1.1 externals ...

diablodale, 2016-03-07:
As a developer, I love puzzles like this. ;-)  Thank you for testing the beta.
I believe this is a setup issue regarding core DLLs. The two issues above may be related, but I do not yet know.

Issue 1 (Max reports error 126 with v1.1 beta)===========
My first guess is that there is more than one copy of dp.kinect2 in the Windows or Max path. And there may be an extra copy of the face DLLs and NuiDatabase folder. Please look in your patch directory, Max 7 packages folder, Max exe folder, etc. Please explore and see if you find more than one dp.kinect2 or more than one set of face DLLs/NuiDatabase folder. It is best that there is only one set of dp.kinect2 files in any given path scenario.

To my knowledge...there is only one core DLL change that I made in the current beta release is that it is compiled to target Windows 8.1 and newer. You are already running Windows 10, so that should be ok.

After you have verified that there are no duplicate copies of dp.kinect2 or face files/DLLs, then please try one thing. Download and install the VS2015 redist package from https://www.microsoft.com/en-us/download/details.aspx?id=49984

Issue 2 (dp.kinect v1.01 reports NuiDatabase folder missing)===========
There is no possibility that simply swapping the two MXE(64) files would change the state of your computer to cause this error. What *is* possible is that the report you are getting is true. You will get this error on both v1.01 and v1.1beta when you enable a face-feature and don't have the needed face DLL/folder in the correct location. Please refer to the setup guide step 7 to verify you have all needed face files/folders in the correct locations https://github.com/diablodale/dp.kinect2/wiki#setup-guide

diablodale, 2016-03-07:
I was able to repro a similar loading issue on a fresh installed Win81 x64 Max 6.10 computer.
Please download and install the VS2015 update1 redistribution package from https://www.microsoft.com/en-us/download/details.aspx?id=49984
Does this resolve the issue for you?

singularmoments, 2016-03-07:
Ive got only VS2013 installed.
should i install the VS2015 update1 redistribution package?

diablodale, 2016-03-07:
Please install the VS2015 update 1 package linked above.

singularmoments, 2016-03-07:
It works, installing both VC_redist.x86 and VC_redist.x64
max 7.1 32b and 64b
Thanks

diablodale, 2016-03-07:
Thank you for reporting this.
I was able to isolate that the auto-parallel feature of Visual Studio creates a hidden dependency on the OpenMP runtime. Installing the VS2015 runtime provides this OpenMP runtime available, so that fixes it for you.

In addition, I have made small code changes in the beta to remove this OpenMP dependency.
Beta version 1.1.20160307.0 or later has these code changes. They should no longer require the VS2015 runtime install. You can download it at the usual location http://hidale.com/shop/dp-kinect2/

In the future, I may re-introduce the dependency on the runtime. However...for now...I don't want to take on the dependency.

closed 2016-03-07 by diablodale