diablodale/dp.kinect2 #19: smoothing value crash max when dp.kinect2 is closed

singularmoments, 2016-03-08:
Step to reproduce:

open dp.kinect2 and start metro
close it and stop metro
send dp.kinect2 _ smoothing 0.5 0.5 0.5 0.05 0.04
crash

it seems that it crashes only after having been started 1 time and only when its closed.
if dp.kinect2 is open, it does not crash receiving the smoothing value

diablodale, 2016-03-08:
Please provide the additional information requested as in https://github.com/diablodale/dp.kinect2/issues/1. For example, I need to know your setup.

singularmoments, 2016-03-08:
Setup
Shenker (Clevo) W705 laptop with 16 GB RAM, 2 x 500GB SSD drive, Intel i7_4790S cpu
Windows 10 x64 with all current updates

Installed Kinect SDK v2.0_1409
Speech and language pack US ES FR
Installed Max 7.1 x64 and it is registered

crash with all versions : dp.kinect2 v1.0.1 _ dp.kinect2beta1.1.20160307.0

Step to reproduce:

open dp.kinect2 and start metro
close it and stop metro
send dp.kinect2 _ smoothing 0.5 0.5 0.5 0.05 0.04
crash

it seems that it crashes only after having been started 1 time and only when its closed.
if dp.kinect2 is open, it does not crash receiving the smoothing value

diablodale, 2016-03-08:
Thank you. I isolated and fixed the bug.
The fix will be available in the next beta release after 1.1.20160307.0
Until then, you can set the smoothing by
1) typing it on the object itself
2) sending the smoothing message before the first open
3) sending the smoothing message any time dp.kinect2 is open and running

diablodale, 2016-03-24:
I've released beta v1.1.20160324.0. It has this bug fix on it. Usual download location https://hidale.com/shop/dp-kinect2/

If you encounter this bug again, I request you reply on this issue so I can investigate it

closed 2016-03-24 by diablodale