diablodale/dp.kinect2 #22: registration problem

noxmox, 2016-03-14:
Win 10
Max 7 32 bits 7.2.1 registered
Runtime 2.0
dp.kinect2 v1.1.20160225.0 

I can't register properly the dp.kinect2 external. 
I followed the instructions and it worked but only with the max 64 bit version. After registering the external in Max 64 bits it works for a while also in  Max 32 bit version, but after closing max the registration disappears and I have to repeate the whole process again.

Thanks!

diablodale, 2016-03-14:
Thank you for the report. Can you provide a series of steps that I can
follow to reproduce this?

Also, please confirm the versions of dp.kinect2 in both scenarios. It can
be easy to have multiple versions in the Max search path. The versions can
be confirmed by looking at the Max console or in more recent versions of
dp.kinect2 by looking at the version attribute.

It may be helpful to enable the @verbose attribute. It may provide more
detailed error codes in some situations.

noxmox, 2016-03-14:
> Thank you for the report. Can you provide a series of steps that I can
follow to reproduce this?

no sorry! I can't see any pattern here.

> please confirm the versions of dp.kinect2

v.032

>  It can
be easy to have multiple versions in the Max search path

double checked

verbose:

dp.kinect2: dp.kinect2 is not registered or has expired. Register your key or visit the software section of http://hidale.com for a new registration key.
dp.kinect2: General access denied error. hr=0x80070005
dumpout1: open 0
dp.kinect2: Kinect has not been successfully opened and initialized
dp.kinect2: Kinect has not been successfully opened and initialized
dp.kinect2: Kinect has not been successfully opened and initialized
dp.kinect2: Kinect has not been successfully opened and initialized

diablodale, 2016-03-14:
dp.kinect2 v0.32 is a pre v1 beta from 2014. I"m sorry, I do not support those betas.

I have noticed that this is contrary to what you wrote in the original issue. In it you said you were using dp.kinect2 v1.1.20160225.0.

While you check your installed versions, please note that the beta version you wrote in your original issue v1.1.20160225.0 expired today 15 March 2016.

I encourage you to download and use the release version v1.0.1 -or- try the current beta which is v1.1.20160307.0. Both of them you can download from https://hidale.com/shop/dp-kinect2/

If you encounter problems with them, please provide similar information that you did above. If I don't hear from you in a few days, I will assume the issue was the two year old beta or that you used the version that expired today.

closed 2016-03-21 by diablodale