diablodale/dp.kinect2 #23: Multiple Kinects v2

noxmox, 2016-04-09:
We are trying to get multiple kinects V2 working on (Win 10, dp.kinect2 1.0.1) but we can't get them at the same time. Using the getusbidlist don't gives us different IDs. 
We can start them independently but not both together. 

Any thoughts?

diablodale, 2016-04-09:
Microsoft does not support two Kinectv2 on the same computer. It is
technologically impossible.

noxmox, 2016-04-10:
Thanks for the info. In that case I would  suggest to remove the "2nd kinect example part" from the dp.kinect2 help file.

diablodale, 2016-04-10:
I'm interested in your feedback.
I'm not familiar with a dp.kinect help file. Do you mean the demo/example patch I include with the download? (fyi, the upcoming v1.1 will have a proper help file and a small set of simple tutorial/example patches).

Hmmm. What is your thinking that drives your suggestion? I did document only-one Kinectv2 in the wiki and Microsoft themselves do the same in their docs. Are you speaking to the yellow box of text in the "more features" subpatch? Or the existence of two dp.kinect2 objects in the patch?

I don't think removing multiple dp.kinect2 objects is the best approach. I want to have example patches with multiple dp.kinect (and dp.kinect2) in the same patch file. There are many usage cases and customers that have this need. For example, it is very easy to have one dp.kinect2 object outputting rgba color matrices while a 2nd dp.kinect2 object is outputting yuv color matrices...both of these running simultaneously. I acknowledge this kind of usage is advanced and will be in additional/advanced usage example like is today with multiple sensors, speech, and face tracking.

With both my Kinect externals, there is logic in them to enable and disable some features based on the capabilities of your computer and the attached Kinect. There are slight variations. For example, Kinect camera control, sensor resolution, etc. My externals will give you appropriate output (like the output from getusbidlist) given the technology being used while maintaining patch-level compatibility so you can switch between sensors as you need.

Microsoft supports multiple Kinectv1 on the same PC. However, they supported...and then retracted...multiple Kinectv2 on the same PC. Both my externals can support multiple Kinects. The feature is enabled or disabled depending on what the Microsoft technology allows. Therefore, the attributes and messages that my externals output *can* change because of this. This is the thinking behind how "getusbidlist" can return multiple ids with Kinectv1 yet on today's Kinectv2 drivers will only return a single id. Nothing would need to change in patches if Microsoft re-enables support for multiple v2. It would transparently start working, as would my externals, as could patches.

What might be clearer is for the help files (not this demo patch) to be slightly more customized between dp.kinect and dp.kinect2. There is little technical need as they are highly compatible. What I could do is adjust text (like that in the yellow subpatcher box) to talk to the feature while also cautioning that certain features may not be able on a sensor/PC and therefore should be queried. Or something like that.

What is your thinking on this?

noxmox, 2016-04-13:
I guess it would be enough to point out that only 1 KV2 can be used on one PC. A comment next to the getusbidlist message... something like "at the moment multiple Kinect is only supported with KV1 (Windows limitation)" 

Btw. I would highly appreciate the direct texture output feature!

diablodale, 2016-04-13:
Ok. I can tweak the text in the upcoming help file(s).