diablodale/dp.kinect2 #27: unregister and register on new computer not working

philhuhges, 2016-08-30:
computer model and name
lenovo ideapad y700

windows 10 pro(64 bit)
Intel (Rcore (tm) i7-6700HQ CPU @2.60gHz 
Memory 8G
SSD 120G

GPU GTX 960M 4GB VRAM

Kinect : latest drivers as of august 30th
MaxMSP: 7.2.4

dp.kinect2 V1.1

Steps to reporduced issue. 

1. install and regiter on new computer 
2. get error message :dp.kinect2: registration key has too many activations; need to unregister on other computers 2016-08-30T17:34:09Z
dp.kinect2: RG: 167772160 RV: 0
dp.kinect2: dp.kinect2 is not registered or has expired. Register your key or visit http://hidale.com for purchase and support.
3. tried uninstalling on other computer
4. reinstall to activate
5. get message error:
dp.kinect2: registration key has too many activations; need to unregister on other computers 2016-08-30T17:34:09Z
dp.kinect2: RG: 167772160 RV: 0
dp.kinect2: dp.kinect2 is not registered or has expired. Register your key or visit http://hidale.com for purchase and support.
6. contact Dale

diablodale, 2016-08-30:
The error message is correct. You can't register because your "registration key has too many activations; need to unregister on other computers".

Each license is only allowed registration on one computer. You can move the license. That's fine and fully expected and supported. I checked the license server. Your registration key is already registered on two different computers. I have no way to name, describe, or locate the computers. The server only knows that there are two of them somewhere.

You will need to visit those two computers and unregister your license. An easy way is to use the help file. There is a unregister button on the "register" tab.

If you did major erasing on those computers (e.g. formatted your drive, reinstalled Windows, deleted files and the registry, etc.), you might have erased the locally-stored registration information.

Easy fix :-)  You can...

1. Register on those two computers using the "register" tab in the help file. This register will authorize with the licensing server and the server will see it has already authorized that computer. It will reuse and re-download the existing registration information. This is similar to a "resync".
2. Unregister using the "register" tab in the help file. The computer has its locally-stored registration information from step 1. It should be able to unregister.

With the above two steps, you will have unregistered your license and now you can move it to a new computer.

If you have problems in the above two steps, please provide similar diagnostic information that you provided above so that I can troubleshoot.

philhuhges, 2016-08-31:
I am fully aware of your policy. 
_Simple Licensing, Simple Cost
The licensing cost of my software is related to the number of computers that have it installed. For example, if you had:
	▪	1 development computer with the software installed
	▪	2 computers for performance with the software installed
	▪	1 backup computer with the software installed
That is 4 total computers that have the software installed. You need to purchase 4 licenses. You purchase only what you need._

-----
I almost have it working. 
Steps: 

-I have 3 computers.

-This license has been has been registered and UNregistered on 2 computer already. 
And following your steps worked this morning. Montreal (CAN) time. 

STEPS THIS MORNING

Computer 1:
-press register "philippe hughes"
-select file dp.kinect2-692.dpreg
-get green registered confirmation
- select unregister
-get a "dp.kinect2" is not registered in Max Console
Perfect ! dp.kinect2 is now unregistered

Computer 2:
-press register "philippe hughes"
-select file dp.kinect2-692.dpreg
-get green registered confirmation
- select unregister
-get a "dp.kinect2" is not registered in Max Console
Perfect ! dp.kinect2 is now unregistered

Computer 3:
-press register "philippe hughes"
-select file dp.kinect2-692.dpreg
- in Max console I get the message error: dp.kinect2: registration key has too many activations;

This is false since I have unregistered the 2 other computer.

Again , what is curious is that following you steps 1 & 2 it did not work yesterday for any computers, 
but worked this morning for the 2 first computers. 
Is it a 24 hours limit time ?

thanks again Dale

diablodale, 2016-08-31:
Are up and working on your one computer? I'm unable to clearly understand.

As of right now 4:47pm GMT+2, there are two computers registered somewhere on your license. No additional registrations would be allowed.
> One was registered on GMT: Tue, 02 Aug 2016 19:47:58 GMT
> Two was registered on GMT: Wed, 10 Aug 2016 15:20:48 GMT

The actions you write above are not appearing in the licensing server. Are your three computers connected to the internet when you are doing all of this? That would be needed to successfully and fully register/unregister/move your licenses. There are complex scenarios that I am supporting which require Internet connectivity in some cases.

There are no 24 hour timeperiods in any licensing code/rules. I also haven't changed any client or server configuration in over a month. I believe this isolates the only changes to be something on your client computers or the steps you followed. To answer one of your questions, the two-step "resync" that I described above in this issue is a major change from your previous tries and would cause different (wanted) behavior.

philhuhges, 2016-08-31:
Hello Again Dale, 
Thanks for your quick response. I will try to be as clear as possible. 

- I have 3 computers. 
- all are connected to internet.
- I have 1 dp.kinect2 license
- they are all unregistered. <- or supposed to be.
- this morning I did your 2 steps process (register/unregister) and it worked on 2 of the computers.
- this process does not work on the 3rd computer. 
----
Yesterday , this process (register/unregister) work on none of the computers. 
Presently, I am in demo mode on 2 of the computers, since I need it to work for the client demo. 

2 computers, 1 for demo and 1 as backup computer. 
When I come back from the client demo Friday, I wish to have it working on 1 computer again. 

Hope this is clearer 

thanks again. 

phil

diablodale, 2016-08-31:
I do not see any questions and everything is working as designed.
Please rephrase your writing as questions.

I do not know any demo mode. What is that?

philhuhges, 2016-08-31:
I mean trial mode

diablodale, 2016-08-31:
Ah. Thank you for the clarification.
Do you have any further questions?

philhuhges, 2016-08-31:
Not @ the moment 
thanks

diablodale, 2016-08-31:
Good to hear all is working for you.
Please remember that you are in violation of both the trial license and your paid license.

* Trials are for evaluation, not doing business work as you are doing for your client presentation. You have greatly exceeded the limits for a trial.
* Paid licenses are for installation on one computer; you can move them by unregistering.

I put great trust in my customers and I ask that they honor that. Licenses for your additional machines can be purchased at http://hidale.com/shop/dp-kinect2/

closed 2016-08-31 by diablodale