diablodale/dp.kinect2 #28: @Facesuausmooth crashes patch

zrthomas, 2016-09-17:
Currently when I attempt to set smoothing params for AUs with face tracking using the @facesuausmooth argument it crashes the patch instantly. This happens when I attempt to set the argument directly into the object box or through an attrui. All other features work normally. Can get AUs out as long as the smoothing arg not set, skeleton data, 2d and 3d face points, etc.

Here is link to example patch https://www.dropbox.com/s/qto158wsthh270o/kinect-facetracking_9.17.maxpat?dl=0

currently running
Windows 10
Max 7.2.5
dp.kinect2beta1.1.20160517
Kinect v2 SDK

best,
Zach

diablodale, 2016-09-18:
I no longer support the betas. Please download the current release from https://hidale.com/shop/dp-kinect2/ and see if you have any problems. This may be related to the already fixed issue https://github.com/diablodale/dp.kinect2/issues/19

zrthomas, 2016-09-18:
Thanks for the quick response. 

More info about my setup
ASUS ROG laptop, 2.6 GHz Processor, 16GB Ram, NVIDIA GTX 980M Graphics Card
Windows 10 x64
Max 7.2.5, registered
latest dp.kinect2 build 1.0.1
Kinect v2 SDK

I checked out the thread you linked, and I am able to set the AU smoothing while the device is running, but still unable to set it while it is closed or in the object box directly.  I also turned on verbose mode and am getting this error message occasionally, which could be an independent issue. "dp.kinect2: data unavailable for request"

diablodale, 2016-09-18:
You are still using an old version of dp.kinect2.
Please download and try the latest version of dp.kinect2 which is v1.1.20160705.
https://hidale.com/shop/dp-kinect2/

zrthomas, 2016-09-18:
Ah, so sorry, that was actually a typo, I'm actually using dp.kinect2 v1.1.20160705.0 Win64. The issue still persists. Can't set the AU smoothing argument from the object box, but seems to work when setting it while the device is running, via an attrui box.

diablodale, 2016-09-18:
OK. Thanks for the clarification. I'll look into the crash this week.

When anything is typed into the object box, Max destroys the old object...Max creates a new object is created...and the startup/config routines for the object are executed. For dp.kinect2, this will stop the Kinect because Max destroys the old object. And when Max creates a new object, the Kinect will still be stopped because it has not received an "open" message.

This is different that you setting the @facesuausmooth values by attrui, Max inspector, or messages. In all three of those cases, the dp.kinect2 object is never destroyed/created.

What I request is to understand what specific series of steps...very detailed...for me to cause a crash in this latest version of dp.kinect2. Please reference https://github.com/diablodale/dp.kinect/issues/18 for an example. I need steps so detailed that step (1) is probably "Run Max 7.2.5 x86". And step 2 is probably "choose file/open and load the sample patch I provided".

With this detail, I can reproduce your crash and then identify the cause. I can also understand which of the two above scenarios (destroy/create -or- just settings values) are occurring...because they run different code paths.

I do appreciate your bug report and I suspect I can make a fix fairly quickly.

zrthomas, 2016-09-18:
Sure thing.
Step 1. Run Max 7.2.5 (64 bit)
Step 2. Load example patcher (kinect_facetracking-9.17)
Step 3. add smoothing attr to object with [0.1 0.1 0.1 0.1 0.1] arguments
Step 4. crash patch
Here's a vid of the process: https://www.dropbox.com/s/gy3ma04dywm1omc/dpkinect2-AUsmoothingbug.mp4?dl=0

diablodale, 2016-09-20:
Thank you. I've reproduced, isolated, and fixed the bug. I should have a new version of dp.kinect2 for you this week.

zrthomas, 2016-09-20:
great, thanks!

diablodale, 2016-09-22:
Hi. I've uploaded a new version which includes a fix for this issue. Please visit https://hidale.com/shop/dp-kinect2/ and try v1.1.20160922

diablodale, 2016-09-29:
Closing this as fixed.

closed 2016-09-29 by diablodale