diablodale/dp.kinect2 #29: Example of playermap in use (not-a-bug)

zrthomas, 2016-09-30:
Hi Dale,

I am wondering if there is any example of the playermap being used to filter an output stream. Forgive me if this exists somewhere and I've overlooked it. I am aware of the @depthonlyplayers attribute which is very useful. However, I am attempting to parse individual users to their own matrix or texture, which I assume is something others have been interested in. Perhaps it is obvious for someone more familiar with Jitter.

Best,
Zach

diablodale, 2016-10-01:
I recommend you ask the Max forums for ideas and support. This is a question about how to use Max matrices; filtering one matrix based on values in the cells of a second matrix, processing them, etc.

Meanwhile, you could send the playermap to multiple jit.op objects and their "pass" operators. Or jit.expr. Or jit.gen and write a gen patch to do it all in one object (this likely performs faster).

Here's a simple patch using jit.op
```
----------begin_max5_patcher----------
845.3oc0Y0taaBCE82oR8c.wOmxhvXyWSpRcOGSSUNIdotCLLvoMsU6ce1FR
ZRqSwKq6FnRETtXfy8v4bsuvyWdwD+4kaXM9dew6adSl7rJxDSLcjIaCLwuf
tYQNswLPeA6gx424Os6XR1FoI9cb4rxJuqU++Iuquml6QB1MJ9RyXTm3mI3c
QEqKJWKyYRyUNba3eTJjM7mX5fnvYA6GWPKLw8+ZMmlu6JUQkKtkKVcSMagr
MeRvnYQYYYIwS8BCSUWF0EKPuSu066aO0VDHerh0dd5D4lBprluwepmu+KiT
AWtXGZMA+8kWn2q1M0YBTSTUOvEKKevJ+P5ie5KYSzblWpIWie+b88xPzImg
NHQt5ppVQRrUNHBJMhhd5nMTZqFAODzHtaxhraxBfg.ixHiSSF5DLYGjriJS
VjUNHDJMBdraxhsZxvIvPfjnnYwAp+FYlLb5IXxNHYGUlLhUNHCJMBYFpi1F
olrT6lLBLDHFELNMYQmfI6fjcTYxvV4fXfzHAYCSSVAqogth8VJbQdYCyFkE
drE.g9XorrHM8jgzawYFFK88kXmwJPnPqKTBAj7BEm71RPIipZ3g1aGAAT+b
gF01fh6Np0rrhIrxUDXbljj8blImEmYOSsgBNgo15j.ipI0PVydDjdlA0zY8
nKRNAYwA0VGrhikUy9IWn.cn20U4zGY0EzJ6xiiUQM9+h5.0pNH+8UTc+WvR
0+pfIqKUKoy61mr91QfoJbK+FY30TjCr6bpX0GuiSVtZUNaOfVqx.Iq9FlfN
O2buCrsv69HoiWfdWt1aFyExya44PqkmgZMgwP1xQ6v7yUEgd8mEx.N8AdEs
1Tttdw16ZGy3sGxVxZjbAUxKE6On3CG0s7kKYhCzYK4MZsmgxCN9CZ2wUrK.
KAdbE4DgE.Ovb6IYD3.SeGw8gKxPEWgCSboeSkCRbAekB8mntegu9C0BMvbp
TwqgOD.KvIfgfGXgNAL3KthSbAX5OgBz.KyIfA+7jXhS.C9x9XmVXAF9xEHm
XrA57jmgZECS1JDMleJB+iQ2l79eBVcsIQqptmU2zckaQjpMy6Jqe4Mw3q51
p82sME6Wytmu8TxZCQqUs8IU87sttssvMwDe0QL2P0l+.Xh4HWK
-----------end_max5_patcher-----------
```

zrthomas, 2016-10-02:
Thanks! found a solution, quite simple.

closed 2016-10-02 by diablodale

diablodale, 2016-10-05:
Here are three examples of using Gen to isolate playermap masks per playerid
```
----------begin_max5_patcher----------
2553.3oc6c01biZqF8ydmo+Gz3OdaRFjDu4N21Y6b+YrcmLXihs1ECbA4jrs
S+uWgDXC1Fjv3X.WFOwNQHLmmijdzijNR4u9oOMa9xn2IoyA+B3KfYy9KdJy
DokkxrhDlMeq26qB7REYb92nrmVQBBVFDs56yeHOKuDExB81RDY42SndA6uT
3tszv.BSb6nhTi8Xq1PCW+bBYESh.KKimLd.3rvL6CrCN6CadZfuVbWqhBdi
5y1H9prJRk5KdrQK+1iVVketQ6XEOXyhjShdqJRxf9KdqDP2nzCRjKnY4rkR
+SQ1fnm1mS4yf8iXhzJlGPSYye.Le+OfuJx5e+SeJ6S9GOLfHanQGHayIxtM
jsMtKjMdhraCYaYA6.YilH61P1lNcgrgSjcaHa7ht3FwXhraEYC6PzHlKtqH
6UQa2RBY6sIF4cAYwouzT.aSRzt0a.dAz0gDe.GzQIfX56DN3OK8Caf9Q1YV
E.gDzOzU7ApB8WhnQ0043dhIk8i.Y8g4cpxV7azP+n2tXCxw0I6CWiSqNU1d
pKxJTsEzWiVRc13jQSpx3rGkFWdzapLNmQowkGsjJiycTZb4Qmnv3vFiRiKO
Z.UFGZbZbxdeUYb2dukgj23O4S6ITXzz2ae3FE82YZK5uywR16WcVbcdYbOS
+c0RBbv97VOVRFd+X9q4GPuvlII4jZAqxijgFPdkjjRiBKWoX1bu33RoOq7M
kUX7sHw2kyCGRiFJSCcHsDxqzhuByCI6kv4eFm72kHIo2s2WGJ6aJxmjDtiJ
.TdpYUOJf1ghsExw8.sjUPkCFxpbE0YyI9TVVg8x0hPhj2nwSKdnzavJ2wx0
YgyR7KGACu7KlDRCiSHo7fv7X4V0gq6SdwaW.64ZhMrZFNNFypW87AOOa95D
peTXFPpVVkkdwi7K.Iaj8dIaRjkPu3yc67pzbFstqlxM1coK8RxJJWJqUuuH
dNKJJn50NbmAjWX4WOlFFdLgxhha3pIz0aZ5tWFwu51F+5EWJ84cgxK+L28A
64TuWOh4YdAA4NTN5I7tWHk2ZhvnxxCjwgqRB83F7lzUIQAAUsa4kd8bWxm2
fXEY+vVLpT8feCz3hJVyOTr6SWSRYGkHyac5QIcpSGdZ6Vl2z+YFYab.2ZNJ
GUlD5JMyK6Ft5EZdjIU8I646ysSHf+2Id.+cD.KB7RPjGCvaLsR5en7cd1do
p0wskziMLeDJVmLPk5GrRCCX4rCZY1gdqJbI8wPY4ciAVswKbMerbAQgqAO9
aY+cR1mB16KFOB+5eDx3EsjWhRHfs6RYfc7N74iEj.dKJ46U91CngjUQ6BYU
8QegDtsqzyqLNAyl3a3PfuqFwPU59+vMJqNTEzXQIFYwIQNbLg.Wz.g.qgPJ
m7ogSkGJ0shvng.T87kgd7EVNyNFp4KzXmu9Yd.FvNWAK2GG1UIigG6D1Zd2
T.m5YLjBFyR3VBtvQVEyTIi0TULmKhwdnletwLIGXMQjpp5YhES7JRRnXG0M
VMF5d6yXD6KmQvNNsiQVLFHjNz8W9nuzmPbGCDhYmCIUeBwYLPH3KmPfxgoq
OgXOFHDTGHD3h1QHVCcBgGQH7xiHzpkADN1CuIq5SGhGTVsQ+ZOlWuZOklQo
rwSV27FH.cVFpgoRi1krpf8yKRAGY.9jTFMb+Ds8k8ApAfUy3FpuOI7Diwml
lMMLxYER+h21CcjtXOafeGmydF731v6CKneN177XGO3vdFahzA51CRnC0A5V
CRnaqUCUiAI10pxt4fD5XcftyfD5l5.c2AIzszA5W2NkJhjpH0qpPmRILwba
yieIlGB0qdA6Hf2nrMfXuDussect4wgacPkSNtMqxIzsPkS0Z7+2GeD7+Bnq
9d15mjx7RXsVvc6ohiWKTnisoiXsPwNHKqrey1wvx0s5Rid9YHpfz340TNU.
N0wfvVvfmUqc7RnqtnA7ie56YqMBCA9bbf2OHIa8h48w8YAGk+6BY0cHn81S
4mahBjyRosbDORINVulCpS1V15xcMoSwKVgAcTHGaIoodqImVnHVG5qofRcg
x4P2olAMUgqqaar.uFb8GBesJHJkbUILyJD1hlIL7PkvpqQeJIfao7N9ZeSZ
rRRSJW1BIzVGmUmyP70fyV5EtNqwYwmeDb3+eKgkDAvFfM+4Uspmzwnkvgnr
casrHZwGYMOA60AliEsdc.4R6Y.Upq05iJQouJQrQDFI4YovTT0g6Ir.MSy.
8jRDkQnomrsgViKUVlaaZoZan4Xz1zSz1P7Xz1zSy1PzXz1zSx1P3Xz1zSw1
vaubzqeXuzswADdrJQI.X6M67sRUwnbsZt+jEC9sxSdwnVZS2ZPIM80jPvm4
OwW8BJK4rVOeE4pT21nwPif1SpTePnRcDV1zyTNASNHmIcpOoS8Icpey0odS
qIuXTRfbmycPKC3hsPT16K9Wh5V6h.CEC0WNy6SJaUoxVk7ETtUmlT1ZGp3U
DNr0jxVy8bI2AC5yH28Ja0vtcDxcuxVMcZGgb2qrUWT6Hj6eksZzNB4NWYql
vrArCwSJasEQCpesmQfxVGwJrrmjPzjvsFyxkaRVqSRC8eEJxUec+O.2yB8i
z4uFhCUw5qb4KOlbd6P3l0ah6zZpLLVSET9o8ib6vOc1+LslJSqoxPaMU90e
sKyGYwLzleFyXodzyti8gOyILqtOAt5SXN2ADlY2mfW8IL66.BC28I.VeBy5
NfvPceBh0mvLuCHLX2m.Y8IL7c.gYz4oLUe9BcOLGySK76zB+NsvuSK76zB+
NsvuSK76zB+NDORib09bAZ.t3FN5B9g2Z+Bs0E6CuE+EZoK1Gdq9KzTWrO7V
9Wn1xzvZDuPjCuE.V6EhD5NhwtyHF61iXraMhw9XtsJdDiczfSwC0dluXjc7
kveg3uv7Wl7WV7W17WMbs8eQm8+HDs9eQiP4bkXJOCWsqQ2D1WvwzwM4ffIS
2HRcS.jG9SobVS7ObCnI.cwxIAhf5rGcUdpNzd4j7QQRuPCBJnnKlVjGd5N1
Mex3bMHktREdLVxNpBCMKSxyPLRLayVu34Ms4syqRfMJuO7qslgwGSKF4Cq5
PcyYji8pVvKm3M0shr8pKP4p89dFunmwC5YKoZEtrThKydAWlJwEtWvEVItP
8BtPJwErWvETItL5EbYnBWlKt43J6LKTMggc5Gfg0BXvaJvL0ojz81yWN53w
GY1O.SoKejc+.Lk97QN8CvTV0G0S0wT50GazO.SsWLT+.LkdKv29VklFZAL6
Apee3sGXnECUFCOTAl0.EXHs.FZw.EX3aeX0G+HqAX39AXp87e6GnqvCkx3w
f8DvTFOFzre.lx3wf39AXp6RB0O.SY7XPX+.LksJgF8CvT5G612nTKX0CdW0
ZVK5Drxmryi1QiBDc7NY73cw3Y1AiMr6EOYmKJv.+s+Q7X5dm
-----------end_max5_patcher-----------
```

zrthomas, 2016-10-05:
Thanks! I was doing it with normal matrix op but this seems more efficient.