diablodale/dp.kinect2 #30: dp.kinect2 Key not Registering

karlskene, 2016-11-02:
Setup
Asus ROG laptop with 24 GB RAM, 256GB SSD, 1TB hard drive, Intel i7-4720HQ CPU @ 2.60GHz
Windows 10 x64 with all current updates
Installed Kinect for Windows SDK v2.0
Installed Max 7 (64 Bit) and it is not registered (part of Max for Live)
Using dp.kinect2

Steps

After successfully using the dp.kinect2 trial I tried to register through the help patch.  I entered my registration name, clicked register and selected my my downloaded key and it still says NOT REGISTERED
This is what comes up in the Max console window:
dp.kinect2: RG: 33556480 RV: 0
dp.kinect2: dp.kinect2 is not registered or has expired. Register your key or visit http://hidale.com for purchase and support.

diablodale, 2016-11-02:
Did you unzip the registration key download? Your registration key(s) are inside the ZIPed download. If you were to try to register using the ZIP file itself, it would fail.

To assist beyond the above, I'll need more information. Please provide:
1. The steps you followed to install dp.kinect2
2. The exact version of dp.kinect2
3. A copy/paste of all the messages in your Max window which will include dp.kinect2 information and error messages
4. I found your order from October 2016. If you are having a problem with another order, please provide the order number.
5. Complete steps to reproduce your issue. Please leave out no step.

Just now, I successfully registered using your license with your registration name using Max 7 x64 on Windows 10 x64 with all current updates. The RG code you provided suggests a core problem with Windows cryptography, or that the registration key is corrupt.

karlskene, 2016-11-02:
The dpreg file is the unzipped key right?  That's what I tried to register
with.

Thanks,
Karl

diablodale, 2016-11-02:
Your registration key(s) are inside the ZIPed download. If you continue to have problems, please provide the information I requested above.

diablodale, 2016-11-08:
closing this issue as customer resolved.

closed 2016-11-08 by diablodale