diablodale/dp.kinect2 #31: "no object"

forrest55, 2016-11-17:
Hello, I believe I have followed the instructions, but I am getting the "dpkinect2: no such object" message. Otherwise, Max 7 is running fine.

Also, when I go to register, I enter "Earlham College" into step 4, but get no response.

This is my second license registered under Earlham College, with the additional payment. I don't know if there is an issue with two licenses under the same name, but this is the name I was provided for order #1652.

Thank you for your help.

diablodale, 2016-11-17:
Hello. Please read https://github.com/diablodale/dp.kinect2/issues/1
I can't help you without that information.

diablodale, 2016-11-17:
Is it working for you now? I'm guessing without needed info.

There is an error "no such object" that is reported by Max itself, not dp.kinect2. If this is the source of the error (I'll know after I receive more info), this means the dp.kinect2.mxe(mxe64) file is not found or loading correctly into memory. This usually indicates something not aligned with the setup procedure or a core computer configuration problem. In my experience, it is overwhelmingly the former.

I recommend you check again https://github.com/diablodale/dp.kinect2/wiki and focus on steps 1-7. Perhaps even do them all again. And remember what is written there, "Be sure to install the 32 or 64-bit version (or both) to match your Max platform version"

forrest55, 2016-11-17:
Thanks, Dale, will get to this in a couple of hours. I do understand that it must be a setup issue or file location issue, usually the case when an object appears in a patch but is 'not found.' Will go through the routine again and get back to confirm assumed success. 

-----

forrest55, 2016-11-17:
Just turning to this now, but here are the specs you wanted: 

I'm on a new HP EliteBook running Windows 10 Pro, v 1607. We're running in 64 bit mode. I downloaded (just yesterday) the very latest version of Max, only the 64-bit version. It's running fine, took it for a quick spin. 

I'm going to trash the download of dpKinect 2 and grab the very latest one from your site and go through the procedure again, carefully. I've already had successful reports on installing Kinect v2.x Runtime and the Visual C++ update, vc_redistx64, so I shouldn't have to do those again. 

I did place the entire dpkinect2 folder into the Properties folder, as I understand your instructions. I did that, all that, already but will trash try again and get back. I have an hour til my next thing...

forrest55, 2016-11-17:
Same response, I'm afraid. I wasn't sure about vc_redstx64 or x86, so I tried 86, but neither is working. 

I have foll0wed this https://github.com/diablodale/dp.kinect2/wiki exactly up through placing the unzipped file in the Props folder. The help file launches, but I still get the 'no such object' message. 

When I fill out the registration tab, I simply type in Earlham College into the registration line. 

I've told my students I'm going to demo this object with Kinect tomorrow in class, so I hope we can work this out. The old machine also has the old v of dpkinect, but since I've paid for a second account, maybe I'll try installing it on the machine where it worked before, it's still here in the room. Is it OK that both accounts are called the same: "Earlham College" ?

diablodale, 2016-11-17:
> dp.kinect2/wiki and

If you use Max 64-bit, you need to install VC Runtime 64-bit and use
dp.kinect.mxe64. This is a Windows issue and the bits must align.

I'm not aware of any properties folder. That might be your problem. You
might be having a simple issue that Max can't find dp.kinect2.mxe64. Like
everything in Windows, your app (i.e. Max) has to know the location of the
thing you want to load (i.e. dp.kinect2.mxe64). You have to put it
someplace Max can find it. My suggestion is in step 7 of the setup at
https://github.com/diablodale/dp.kinect2/wiki. If you changed that folder's
name, folder location, or change the default Max setting for file
preferences, you'll have to move it elsewhere that Max can find it.

I'll continue to need to know all the information I requested. I lot of
information is still needed.

forrest55, 2016-11-17:
OK, almost there. I created a specific file path to the dpkinect folder, and now the object is found. Looking good. 

Toggling it on is now bringing up the please register request, and I am typing in Earlham College, as per instructions from the payment response. 

But the Kinect is not activating, so now we're at that stage. What do you suggest? I am plugged into the USB port with the same setup, and USB/Kinect devices, as last spring. So this seems to maybe be a registration issue now. 

-----

forrest55, 2016-11-17:
Yes, so this is the same problem now that we had before. I am now getting the "dp.kinect is not registered or has expired" message, even though I am entering Earlham College at the prompt. I pad for a fresh license yesterday, so I should now have two registrations with you, both under the same name. 

I am also getting: 

dpkinect2: General access denied error. hr-0x80070005 

and 

dpKinect2 requires Kinect20.Face.x64.dll 2.0 or greater. But as I am not wishing to use face recognition yet, I don't know if this is a problem at the immediate moment. 

This happened before. 

----- On Nov 17, 2016, at 2:09 PM, tobeyfo <tobeyfo@earlham.edu> wrote: 

OK, almost there. I created a specific file path to the dpkinect folder, and now the object is found. Looking good. 

Toggling it on is now bringing up the please register request, and I am typing in Earlham College, as per instructions from the payment response. 

But the Kinect is not activating, so now we're at that stage. What do you suggest? I am plugged into the USB port with the same setup, and USB/Kinect devices, as last spring. So this seems to maybe be a registration issue now. 

-----

BQ_BEGIN

— 

BQ_END

diablodale, 2016-11-17:
I looked at my store and see that you haven't downloaded your new registration key you purchased. You must have this key to register dp.kinect2. The registration name and key are paired together. Please see step 3 of https://github.com/diablodale/dp.kinect2/wiki#register-dpkinect2

The face error is there only because the patch you are loading tried to use the face features yet the face functionality wasn't installed as in step 8 of the Setup Guide. You can ignore this error or remove the face features from your patch.

If you continue to have problems after you use your registration name and downloaded registration key together, please refer back to the information I requested as I'm still missing the bulk of information regarding specifics on your setup procedure, what steps/buttons/clicks/patches you are following, etc. Also, the actions you took for "a specific file path to the dpkinect folder" as that seems very unusual to me.

forrest55, 2016-11-17:
It's probably just this, checking it out now ... 

-----

forrest55, 2016-11-17:
I'm sorry, I see a prompt for the registration name, "Earlham College", but not for the registration key, which I take to be #dpswreg-9. Where/how do I enter this? I mean, I've already purchased it, and got the attached message, screen shot. In fact, I received four of these in a row in short succession.

forrest55, 2016-11-17:
Almost there. Now, for the first time, entering "Earlham College" into the register box brings up a dialogue box to upload a file. I chose "LICENSE", but that didn't seem to have any effect.

forrest55, 2016-11-17:
OK, good news is, on the short term, like tomorrow, the 14-day trial is working and I am getting image, so all is now functional. I'm heading off to rehearsal, but let's see if can clear up the licensing thing today. I won't probably be back online now for about 4 hours, unless you get back in the next 15 mintues or so. 

Great to see signal though, so in general all is functioning on the new machine. Yay.

diablodale, 2016-11-17:
All seems in order.
If you choose to use the help file with dp.kinect2, it has a registration alert and tab.

You have chosen to do the trial; following step 2 "register trial"

When you want to purchase and use a license, you can follows steps 3-4 on that same tab labeled "purchase license" and "register". You did step 3 earlier this week. In those emails, you received your registration name. You also received a link in those emails to click on to download your registration key. Download your registration key with the download link you received in email. There are instructions in that email regarding this procedure.

This download is a ZIP file. ZIP is a de facto archive standard. They contain files and folders within them. The ZIP file itself can not be used directly. Instead, your operating system (Windows) or add-on applications can open the ZIP archive for you to access things stored inside. You will need to use your OS or an add-on app to extract or "unzip" the contents of that ZIP file you download. It will contain your registration key(s) as you may have purchased multiple keys for multiple Max plugins in your order. These filename of these keys ends with ".dpreg".

Now back to the register tab. Type your registration name into the field which is labeled "When you purchase a license, type your registration name here. This is usually *not* your email address. Then click register."

Now choose the ".dpreg" key file that you unzipped above.
You should now be registered.

In summary:
* Look for your order emails. Those emails have instructions writing for you to...
* Download your registration key using the link in the email
* Download dp.kinect2 using the link in the email
* Take note of your registration name in the email
* And provides full setup instructions using the link in the email if you want them

Once you registered, if you have any feedback on what was not clear, I'm open to hearing it.

diablodale, 2016-12-02:
No reply, closing this as resolved. I have already updated the setup documentation to provide more details on steps to register based on conversations above in this issue.

closed 2016-12-02 by diablodale