diablodale/dp.kinect2 #33: Problem with error message

dolphinazz, 2017-02-10:
HI 

I cannot activate 'dp.kinect2' because of the message below.

"api-ms-win-crt-runtime-l1-1-0" is not found..

I am not sure this error message is from window system or 'dp.kinect2', but I really need to fix this message. 

# version
- MacBook Pro(Retina, 15-inch, Mid 2014), using window 8.1 through bootcamp
- Max 7.301 (64bit)
- dp.kinect2

I really appreciate anyone can help me.

diablodale, 2017-02-10:
Hello. Please read https://github.com/diablodale/dp.kinect2/issues/1 and provide all the information requested.

I caution you that this is a Windows and/or setup error and I do not support Windows. Windows support is available at https://support.microsoft.com

diablodale, 2017-02-13:
Did you resolve this issue? If not, please provide the information I requested from that link.

dolphinazz, 2017-02-14:
Thank you for your concern. I solved my problem through 'window update'

Thank you again~

diablodale, 2017-02-14:
Good to hear. :-)

closed 2017-02-14 by diablodale

dolphinazz, 2017-02-16:
Hi, I am happy to use kinect2 with my laptop, but i want to use mac system
because of many useful things to me. So I want to buy kinect 1.

Is there any points I should know before buying it?

Now, "jit.openni" the external is not activated in my laptop.

I will use kinect 1 in Mac OSX.

Any version of the product ' kinect 1' is ok for Mac OSX?

Thank you.

diablodale, 2017-02-16:
I'm locking this issue to keep the topic on track.

You can find information about my solution for the Kinect v1 sensor at https://hidale.com/shop/dp-kinect/
Microsoft requires the Windows operating system for both Kinect sensors. I recommend you investigate bootcamp if you want to use your mac laptop.