diablodale/dp.kinect2 #34: Registration Stopped Working! General access denied error. hr=0x80070005

duffield, 2017-02-22:
My registration simply stopped working! Do not have registration on another computer. I did recently upgrade my OSX so perhaps this is the issue? I also have not unregistered my license.

Max Console Errors:
dp.kinect2: dp.kinect2 is not registered or has expired. Register your key or visit http://hidale.com for purchase and support.
dp.kinect2: General access denied error. hr=0x80070005

My Computer:
Macbook Pro 15" (Mid 2015)
2.5 GhZ Intel Core i7
AMD Radeon R3
Windows:  10.0.14393
Max 7.3.1 64-bit 
dp.kinect2-v1.1.20160705

Steps I tried:
Re-register with the key I had before and logging in. No success.

Let me know how to troubleshoot this! I have a tight deadline!

diablodale, 2017-02-22:
Hello. Please read 'how to submit an issue' here https://github.com/diablodale/dp.kinect2/issues/1
I will have more info on which to help you troubleshoot.

Meanwhile, there are improved error messages in the newer versions of dp.kinect2. You can download the latest version from https://github.com/diablodale/dp.kinect2/issues/1. The entire copy/paste of your Max console will definitely help as it contains lots of diagnostic information.

I recommend you use the included help patch for dp.kinect2 to register during our diagnostics. This will reduce the possibility of patch errors + I have enabled the @verbose option to provide more diagnostics.

diablodale, 2017-02-23:
Checking back on this. Did you resolve your issue? If not, please follow my suggestions above so that we can continue diagnosing.

duffield, 2017-02-24:
Apologies! Was working on a different project yesterday :)
I have outlined more detailed troubleshooting.

My Computer:
OSX Sierra 10.12 (recently upgraded from Yosemite) 
Macbook Pro 15" (Mid 2015)
2.5 GhZ Intel Core i7
16GB RAM 1600MHz DDR3
500GB SSD
AMD Radeon R9 M370X 2048MB **note I made an error when I posted in the first post of this thread
Windows: 10.0.14393
Max 7.3.1 64-bit (registered)
dp.kinect2-v1.1.20170110
Kinect Runtime 2.0
Radeon Graphics Card Drivers updated

Issue:
I had registered dp.kinect2. Has been running fine since November.

Steps Troubleshooting 1:
1. I tried starting up Max and running my original patch (which was registered in November I believe, and was registered using the dp.kinect2.maxhelp).
2.  Max Console reports the following when trying to start the Kinect:
dp.kinect2: dp.kinect2 is not registered or has expired. Register your key or visit http://hidale.com for purchase and support.
dp.kinect2: General access denied error. hr=0x80070005
3. Close Patch. 

I did not register this key on any other computer. I also did not unregister the key.

Steps Troubleshooting 2:
1. Open up dp.kinect.maxhelp 
2. Attempt to "re-register" with the .dpreg file in my Downloads folder. (link in email says key has expired so I did not download a new copy of it).
3. When trying to register Max Console Reports:
dp.kinect2: RG: 33558528 RV: 0
dp.kinect2: dp.kinect2 is not registered or has expired. Register your key or visit http://hidale.com for purchase and support.

Steps Troubleshooting 3:
1. Updated dp.kinect2 to most current version.
2. Repeated Troubleshooting 2 steps. Same results.

Steps Troubleshooting 4:
1. Attempt to use @verbose 1 in dp.kinect2.
2. Console logs: 
dp.kinect2: dp.kinect2 is not registered or has expired. Register your key or visit http://hidale.com for purchase and support.
dp.kinect2: General access denied error. hr=0x80070005
dp.kinect2: dp.kinect2 is not registered or has expired. Register your key or visit http://hidale.com for purchase and support.
dp.kinect2: General access denied error. hr=0x80070005
dp.kinect2: Kinect has not been successfully opened and initialized

Final message repeated with each metro bang (as expected). This makes sense as the Kinect won't run so verbose being active shouldn't output any info?

Questions:
1. Should I try un-registering it and re-registering?
2. I notice since I upgraded my Mac OS from Yosemite (forget the exact Version number) to Sierra 10.12, my boot camp and Max has been unbearably slow, especially Max (i.e,. sometimes takes 10 seconds to make an empty new object). I did not do a clean install, just an update. I am considering re-partitioning my computer and re-installing Windows 10. However, I want to ensure whatever is happening with the license is resolved in case it thinks that it's a "new computer / registration". I know you have a "1 license for machine" policy.

Let me know if you need any more information.

Thanks again!

diablodale, 2017-02-24:
The info you've provided in step 2 is that either your registration name is incorrect, your registration key is incorrect/corrupt, or there is an unexpected problem in the cryptography system of Windows. I'm not able to discern between these.

I see you made two license purchases, thank you. One of the licenses has no registrations. The other license has been registered on two computers. These two computers were both registered on the same day about 10 minutes apart. This could be two physically different computers. Or it could be related to a hypervisor/virtualization if such makes the same computer appear significantly different.

Some ideas:
* Most often this error in step 2 is an incorrect registration name. One accidentally uses their email address, or casual name. Instead, the registration name is a very specific string which is sent to everyone in an email. It and the registration key are a pair. This name must be typed exactly like it appears in the email including capitalization, spaces, and complex characters. **Caution**: I looked in my store order records. Your registration name is different on your two purchases. Please look very closely. You need to use the matching registration name->registration key. One can't swap them. And one must type it exactly as from the email.
* Sometimes customers select the ZIP file as the registration key. That will not work. One needs to unzip the file and select the .DPREG file within it.
* On rare occasions with errors like this, a customer refreshed Windows or used Windows update; then after doing so the registration worked. This suggested on those scenarios some problem with the core Windows cryptography system.

You can unregister if you want. It should be harmless.

The performance issues (10 sec for an object) in Max are probably a separate issue. Such doesn't touch the dp.kinect2 codebase. I do not have experience with bootcamp. Only rumors that some customers do use it and their (good or bad) experience using it is bound by the hardware. The Microsoft drivers for the Kinect2 sensor have specific hardware requirements for the USB controller and graphics card. You can use their verification tool which I link to from step 1 of my setup guide https://github.com/diablodale/dp.kinect2/wiki#setup-guide

My goal is to get your install working. Separately...how you got to this unregistered state is curious to me. I don't have any answers as the diagnostics that may have appeared in the Max console when that event occurred are lost now in the past. Your OS upgrade may have something to do with it if Sierra significantly changed the state/not of a hypervisor in bootcamp or significantly changed how your hardware appears.

duffield, 2017-02-24:
Ah, I did not realize that I had two different Authorization names! That fixed it! 

I think the update unregistered me somehow but yes it works now!

As for the performance issues, yes they are separate issues! I just wanted to note this as I think the OSX update led to Bootcamp having general problems which as previously stated, it appears to have unregistered me. The ultimate issue was my wrong account info. Prior to the update I had no problems with performance or the license!

Thank you for your help!

diablodale, 2017-02-24:
Glad to hear it is resolved.
Happy patching!

closed 2017-02-24 by diablodale