diablodale/dp.kinect2 #35: Registration stopped working

emm-ess, 2017-02-28:
Setup:
OS: Win 8.1 Pro x64 with all Updates
Processor: Intel i7-4770K
RAM: 32GB
Kinect-SDK: 2.0.1410.19000
Max: 7.3.2 x64
dp.kinect2: 1.1.20170110

Step 1:
. start Max
. open project which also opens patch with dp.kinect2
-> Max console output:
```
viddll.engine version: 1.0.7
dp.kinect2 v1.1.20170110.0 Win64, Copyright (c) 2014-2017 Dale Phurrough. This software comes with ABSOLUTELY NO WARRANTY.
-All rights reserved except for those documented in the dp.kinect2 terms at http://hidale.com and granted by purchase/agreement to any entity listed as registered below.
-Compiled with Eigen 3.2.8 licensed under Mozilla Public License, v2.0. Source code & license at http://eigen.tuxfamily.org/
-BY USING THIS SOFTWARE, YOU ACCEPT THESE TERMS. IF YOU DO NOT ACCEPT THEM, DO NOT USE THIS SOFTWARE.
dp.kinect2 is not registered or has expired. Register your key or visit http://hidale.com for purchase and support.
kinect performance caution: power setting (USB selective suspend) should be disabled
kinect performance caution: power setting (Minimum processor state) should be 100%
kinect performance caution: power setting (Intel Graphics Power Plan) should be maximum performance
```

Step 2 (trying to reregister):
. open dp.kinect2 help
. open register tab
. enter registration name ("Marten Schälicke" copied from mail and in a retry of the step manually entered)
. select license-file (dp.kinect2-595.dpreg)
-> Max console output:
```
dp.kinect2: RG: 167772160 RV: 0
dp.kinect2: dp.kinect2 is not registered or has expired. Register your key or visit http://hidale.com for purchase and support.
```

Step 3 (verbose output):
. add @verbose 1 to dp.kinect2 object in patch from step 1
-> Max console output:
```
dp.kinect2: dp.kinect2 is not registered or has expired. Register your key or visit http://hidale.com for purchase and support.
dp.kinect2: General access denied error. hr=0x80070005

```

Step 4 (try trial):
. open dp.kinect2 help
. open register tab
. click "register trial"
-> same output as in step 1

current changes on PC (on the weekend):
. after a long long time installed Windows Updates
. updated Max
. updated dp.kinect2
. installed dp.kinect2 face-tracking support
-> but I have in mind that it still worked after these changes.

changes today:
. restructured & moved directory for Max-Externals (put dp.kinect2 in the directory "~/Dokumente/Max 7/Packages", included a volume/harddrive change)
-> hadn't checked if dp.kinect2 worked before that

. otherwise no change in domain-name or user-account

additional info:
. license bought on 1.3.16

hopefully that is everything needed.

Thanks in advance
marten

yaki4, 2017-02-28:
I have the same issue since the update 7.3.2 of max

diablodale, 2017-02-28:
A workaround is to turn off networking at the time you are registering. For example, disable your wifi or unplug your ethernet cable. It will grant a temporary registration. Then, you can enable your networking.

This is related to the global failures happening today with Amazon Web Services. Only customers trying to register a new install of dp.kinect2 would encounter a problem. Or for a subset of customers that upgraded dp.kinect2.

I planned for this (AWS failing) so the workaround I provided above should work for you. :-) Please do reply that it worked or not.

emm-ess, 2017-02-28:
Thanks for the fast reply. And it worked.

yaki4, 2017-02-28:
Same here. 

Thanks for the fast reply.

It works fine.

closed 2017-03-05 by diablodale

diablodale, 2020-06-30:
The just released v1.2 of dp.kinect2 includes client-side changes to better handle the specific catastrophic AWS failure that occurred on this date. (Server-side changes have been in place for a few years.) You can download it from https://hidale.com/shop/dp-kinect2/