diablodale/dp.kinect2 #36: Face models

Sonopopee, 2017-03-04:
I can't download dpkinect2 face models, is it normal or it's a bug ?

Thank you for your reply

diablodale, 2017-03-05:
I experience no problems downloading in my testing. There are several ways to get that download and they all work well in my testing. For example:
* download section https://hidale.com/shop/dp-kinect2/#download
* step 8 of my setup guide https://github.com/diablodale/dp.kinect2/wiki#setup-guide
* in the help patch on the _face_ tab

If you continue to have problems, please read how to submit an issue here https://github.com/diablodale/dp.kinect2/issues/1 so that I have information on which to assist you.

Sonopopee, 2017-03-06:
Hello,

Thank you for your reply.
It works with safari on a mac for Me.
And all the links failed for me with a pc and firefox...Maybe it's a 
firefox bug.

Best

Max

---
L'absence de virus dans ce courrier électronique a été vérifiée par le logiciel antivirus Avast.
https://www.avast.com/antivirus

diablodale, 2017-03-06:
If you continue to have problems, please follow the instructions I provided at the link above.
I'm going to close this issue as not reproducible for now.

closed 2017-03-06 by diablodale