diablodale/dp.kinect2 #38: Heartbeat detection

aartcore, 2017-06-07:
Hey Diablodale,

I was wondering if there are any tools in the dp.kinect2 to detect the heartrate.
I saw some videos online where they measure it with the kinect one, like this one: https://github.com/dngoins/Kinectv2HeartRate/blob/master/README.md

thanks,
aart

diablodale, 2017-06-07:
I do not know much about heart rate detection. A saw a video demo from Microsoft where they detected heart rate. And then another post by a software developer who made a prototype of heart detection. https://channel9.msdn.com/coding4fun/kinect/Kinecting-to-your-Heartrate

If I remember correctly, both approaches used the IR camera and the RGB camera to look for rhythmic variations in skin color. I have not heard anything after that. Google tells me there are researchers writing papers on similar ideas

https://www.google.de/search?q=david+kinect+heart+rate&oq=david+kinect+heart+rate&aqs=chrome..69i57.4690j0j7&sourceid=chrome&ie=UTF-8
And a video https://www.youtube.com/watch?v=ONZcjs1Pjmk

All the raw infrared and rgb data from the Kinect is available to you with dp.kinect2. You can get that data and then with Max implement any data/math processing on them to achieve these kinds of heart beat solutions.

If you develop a solution, I would very much enjoy watching a video or demo of it. :-)

aartcore, 2017-06-08:
Thanks for the links, it's really interesting stuff! 
Haha! Yeah i would love to make a demo, but unfortunately i think waiting on the next kinect camera where it's probably already implanted will be faster :-)

closed 2017-06-08 by aartcore