diablodale/dp.kinect2 #39: Two dp.kinect2 objects with @faces 1 crashes

aartcore, 2017-06-16:
Hey,

Just want to report a crash/bug, looks like that I finally found how to prevent it crashing.
It looks like if i've two dp.kinect2 objects with @faceprop 1 @faces 1 it gets a conflict running for like 15 / 20 minutes (with (multiple) people moving in front of the kinect-camera).

I stripped down my patch to search for the crashing element. When I run this patch it crashes in 15 / 20 minutes. Now I removed the in first (left) dp.kinect2 the face arguments and now it still running after a night testing (in combination with looping pre-recorded material in Kinect Studio).

I attached also the crashreports which I got: Or with a fault in the Kinect20.Face.x64.dll or with KERNELBASE.dll (most of the time).

If you want more info, i'm happy to help!

dp.kinect2 v1.1.20160922.0 Win64
Max 7.3.3 (64-bit)
Windows 10 - Mac Mini i7 3.0 GHz

crashreports:

Faulting application name: Max.exe, version: 7.3.3.36780, time stamp: 0x58b6c502
Faulting module name: Kinect20.Face.x64.dll, version: 2.0.1410.19000, time stamp: 0x54441cc5
Exception code: 0xc0000005
Fault offset: 0x00000000000a4335
Faulting process id: 0x120c
Faulting application start time: 0x01d2e5ae775e1bd0
Faulting application path: C:\Program Files\Cycling '74\Max 7\Max.exe
Faulting module path: C:\Users\wincore\Documents\Max 7\Packages\dp.kinect2\externals\Kinect20.Face.x64.dll
Report Id: f6895360-25a9-4372-aa6c-173bf0c1c804
Faulting package full name: 
Faulting package-relative application ID: 

___________________________________

Faulting application name: Max.exe, version: 7.3.3.36780, time stamp: 0x58b6c502
Faulting module name: KERNELBASE.dll, version: 10.0.10586.873, time stamp: 0x58d9fd3a
Exception code: 0xe06d7363
Fault offset: 0x0000000000071f28
Faulting process id: 0x1730
Faulting application start time: 0x01d2e5b4062a9b80
Faulting application path: C:\Program Files\Cycling '74\Max 7\Max.exe
Faulting module path: C:\Windows\system32\KERNELBASE.dll
Report Id: 5944e43e-4746-49f6-85c4-023306ecd904
Faulting package full name: 
Faulting package-relative application ID: 

test-patch:

<pre><code>
----------begin_max5_patcher----------
6349.3oc6cs0aqijb9Y6eEDBA4gM9nsuxlLHObxrAXPdXAVjfDffAAFTRz1b
NzjJTTmKyf8+d5KjTjh2Zdo0EObm8Xa0TRrpup5pqtXUU+6O9vpMwe2+vJq+
Yqew5gG98Ge3A4PhAdH60Or5cuuuMz6f7ssZa76u6Gkt5I00R8+dpb7iG7Sr
92+2xGO536AQg9oxODLav8doaeKH50mS72lpto1D20.WWWl8SVHWxZvSVTl3
mHvZf0+6ous3io4ecfrQC1Iuywa90Ogf423Co+HzWN9Jw.+8GeT7imzj632n
M9ICkIntz0NXLFhOwD.ISfZkIPEecIdu6m5m7rej2FEo2DCBxIJ02Q5O16qt
4qV8j0pMdQutp3NMML3c+CG7d0ulD9Eus9VHq2NFr6vWB8OlXAVyYYFhx+CD
.6hP7+.BY1DViHHpaDjiWXNfQEXHtGzCVGgfNsiPyDxr2KxObnJGLGzZFmov
TjfofXnfEgZnemDeLZm+tVzHfPTNorINYGWqsDcThOyF483cUzs3JLYJaqyG
ZyqaiCiSTjctr8oBg6SER2B5deR793jzf3H02D1cDn5uFjtd+2Bh1E+sUEeu
9G3FY7Deyk0RPDA1gQtJKDfheUFI0aJKyY8IgRaeUMLesrDfz4bRtV2Xr.4+
M92csIeBPJdu0m4+6OY84u5EZQJL48RbT5gfeSdug3Bwoty8vXITPbsWCQRU
SE7hGDVfKSMQbSZxK8ulD3E1JJwYpme2KMI36Y307XCucDbareB2HFzJLlau
b.vGre3yFjAe.zHfO3MC701B7Q9oeKN4KeZSRvtSqNnMJYKWVj.kKtCyVdbP
qwCAyyZ75auIWzBrUKNoHZxXr2j8UQwJ9mHWtCBcWyW8TasDSXuY.nACHXAa
kIyIgFPGXIfcjnA9xY8cmeHedM.znICzfs3h.z0zLmdIk84cPN7fcGgMi4zQ
wV.qvXucxay..Kndfkx5gMX3fEhdsAqQsyBHhUv7TaSrwBhyEaiEcrvrZYLK
h0127RrnPjEAQlIEnLiMXlxJC1US.rLHAtYVdtscmwsb91uMzscoPFkhEYLy
pvFeOWs4Nxq9oVG9RPjZKKCMpCfRQc.JYb99AFrGINrUl1PpPbaAWMtc2.YP
IqgFtj0gZbIaZ7qhsfNNVKSpgH4+rGVSSqgNs6hU.WIzrFB2aIBwxm14k5MV
TgHMtoD6cHvIMv4.cMj8TgCVsFrAIEphHwCEX.23YPn+W8SNjEufL94gUd62
WZ3GJ8QD.2upBIA6ohgBhTCgKFJw+qA4edRwndIbbJkCRGSTz22crWc5qIdm
eRzw.IonFjKBeL+KrX4ChzxfJrdttx4RN1zSXqHvIgwa+RkH0vgw89QAQk8k
[... 96 lines cut ...]
Ap13g.MP1fVbjVKTUvYG1Rw0QhwXxn0SumIBZssHS4iwJs.rNV6ARjIRJTcr
LfcG9pOT5HfLSCQkNB1lLBoToxQH1HPH6qHBIV4Y+2Bh1UjBHqJe1oVhQOYO
Fm0imIfQXLRsNBQcN7lYMRr7Mhp+JJHxpdVNXd8gIdu0m4+6OY84u5EJ1rEZ
lTdTfAlkUD484bWSPwsiSLs.fYmFXnxt9N04bxkxTVwqele0+LN6wZEelPpJ
Ma8Yc0+iQ784HNWja2apxSZaEaUNKfX83qvo44s1C2qXoqUzsbuauAm35koF
qKbsnyVtyKuyee5aYcdYAEmuQlp8jY4eW8xp1sLL6aPb5xuOz6G7MjagZrcM
KuX1WW9kEu0Mw69Qs18748.Z48+vIGkD4E713ip3zf0cSEnAuOLk5BT0TVI.
0uZe6Aj5JEvwr5WSM34VlTJub098oRy377mNS+nVdS6Tokaz1oyRotwRMcPs
uSDctSf43NUIgzaKO6ms6Dpm6DrRAvWKcy4V2syJ2cDyUVRSNL49+UubNHw9
fcHYt.Cbe2opEnaMzHKFPLUixzEIwC4KmH0UsoR1cEnLsaDRG.GNC.tsN2nt
PakGZPTtqY4uXhzEcxDV9g9QQWXll8hoRYzISYThYnLM0NKErnJErjQuwn9U
hTsu7L4lSYGVqV8QilNq1L3L37WhNFLwyvzWrqF2HZmPunRvxJALUOcF5xJJ
zs5i.QkmnOYcVjN5NX2Y.ppVTbFT3izw3MZNVsDogOJT6Y5F0qVFcFtQPcDR
yAzUsbk0nSCL96jNtr1uwQ9FcWynEmtCxWM21GgZ37ECMSXReJtDvEB7QZrz
DV09rwRieXU2ytq0PU6oxa+dwyWN61IIR9tq+U0VzYOIeYPj5kxsftJw+qA4
ue4VAW4kv2QYJe6jGSTac66Npnyr58X9MN5XP18V9ThOqrQUfUOkK5YOr7e+
w1R3i7mubyI5wXJKzQVRniL0NZNsNZOkNZIcNxef2O8X6OQ+pOM+xZtsVlms
IFFdAPMSjV4R3TOhSmx1bxDWWkmYyj4XJIywUNlSl4pW1kMyRis13lSBDcqS
f3acBjbKSfkKyQsoPsJDsYkBQ27TH9lmBI2vTX4xJTaBTqRIblHvJkLn9TXO
4C+rRbnaYhCeKSbjaSh6rRzSaxSqxxalnwpkeWyjXOkyyHKkm4kAP26L.9dm
AH26L.8djApV1Z5aATmRUalnte9FbY2Zkal1zlVERwbRfnacBDeqSfjaRBrV
4cYjH7oaYbMxR35CQL9pUVVZDfJMKEqQUFVSlgpWtUMyQCtDqFW4UML9g+Y9
6O9+CYNw2KA
-----------end_max5_patcher-----------
</code></pre>

diablodale, 2017-06-16:
Thank you. I have a few requests to better understand, please.

This may be a known crash in Microsoft's drivers. I write about this in the 2nd bullet.
https://github.com/diablodale/dp.kinect2/wiki#known-issues-limitations-report-problems
There was some forum chatter about it at https://social.msdn.microsoft.com/Forums/en-US/10425cc3-d732-4163-85a5-7f6c4ae983c0/kinect20facedll-crashes-in-repetitive-testing-of-3d-model-features?forum=kinectv2sdk#6562810d-05f1-434f-9357-1beb48f0c957 and some private emails between Microsoft and I that I can't share.

The crash I isolated and write above above appears similar to yours. Mine occurred when I was looping prerecorded material, not in a live scenario. This known crash occurs when the Kinect2 HD face features are used. In your example patch, you use only one option that activates the HD face codepath: @facecolors

I reviewed the crash details. Good bug report! :-) The crash in your report is caused within Microsoft's driver code Kinect20.Face.x64.dll.  You can see that on the Faulting module name line in your report. The type of crash is Exception code 0xc0000005 also from your report. 0xc0000005 is the Windows OS error STATUS_ACCESS_VIOLATION. This means that Microsoft's code in Kinect20.Face.x64.dll is trying to access a place in memory that doesn't exist, is invalid, doesn't have security permission, etc. There is definitely a bug in their code.

It may be possible that my code is sending something invalid to their driver and they do not gracefully handle invalid data. In this case, both Microsoft and I have a bug.

I tried many approaches in 2014 with the crash I isolated. The forum writes about that. I found that I could prevent dp.kinect2 crashing using overly strong error handling in my code. But it wasn't very useful. Why? Because when I did that, the Kinect2 drivers and subsystem still crashed. When the crash cascaded upward to my code, I was able to catch the crash to prevent dp.kinect2 itself from crashing. That left dp.kinect2 in a good state but the entire Kinect2 drivers/subsystem in a corrupt state. It wasn't helpful. This kind of behavior within Microsoft's Kinect2 code suggests a bug that I might not be able to control. Still, I'm happy to continue investigating.

I suspect (still unclear) that the crash you are experiencing and the known crash I isolated are the same bugs. It help further isolate, I have two requests of you:

1. In your first paragraph, you write "with multiple people moving in front". Did the crash you isolated occur with two dp.kinect2 with live people for 20 minutes? Or, was this a recording of multiple people that you looped for 20 minutes?
2. If you remove/disable @facecolors on the dp.kinect2 (top level of your patch) in your example patch, do you experience the crash? And if it does crash, is this with live people or looped recording?

aartcore, 2017-06-20:
Thanks for your reply.
The crash occurs also with the live stream, but not at a specific time, looks like it depends how much activity there is the sooner it crashes. Unfortunately it's hard to get some test persons who want to stand for the camera and be active for like 20 minutes... The crash was especially in situations when i want to show someone the system and there is more activity then normal like sometimes one or two persons do short movement tests in front of the camera.
With a recorded loop of 3 people moving the crash is about 15 / 20 minutes.

With the @facecolors turned off it still crashes (with prerecorded material). May be i can test in a live situation, but for now i don't use the facecolor information because it's to slow. I now use the solution in the patch to get the skincolor.

I'm trying to more test with live people. I just made a sort of a human doll which i can move with pulling ropes so i can test more the real live situation. The doll is recognized by the kinect as human so that 's it's already a first step :-)

diablodale, 2017-06-20:
I just remembered something. A caution my Microsoft contact told me. They said the recordings and the recording subsystem are not playing back only an RGB image. They said that there is preprocessed Kinect data in them. This can be understood through seeing the recording tool allowing different types of data to be recorded and played back. For example, if you choose to record the RGB and depth, but not the skeletal data, then on playback it will not reprocess the RGB and depth to derive the skeletal data. That skeletal data has to be initially recorded.

They said that there is a bug (suggesting there was more than one bug involved) in playing looped recorded materials. They cautioned me to always have the people enter and exit cleanly. Meaning, a person should walk into frame, do all the activity you want, walk out of the frame. They said it was related to how the code didn't handle the sudden change in data on looped recordings. I remember them describing a scenario with faces. How if the enter/exit isn't clean, that when the cut/loop occurs, it would (to their code) seem like a face suddenly disappeared, a new face appeared 1 meter away, and the new face has internal data identifiers for data that no longer exists, plus the skeletons previously/instantly recognized become unstable.

After that, I remember that I re-recorded some material, and then edited the recording to keep only the section of someone someone cleaning walking into the frame, doing things, and then exiting the frame. Then after that, I had no crashes except with the HD face. I think the longest test I did was five days of that looped recording.

Can you edit your recording to have a clean enter/exit? And then see if you have a crash without facecolor?

diablodale, 2017-06-20:
And one more question to help isolate. When you disabled @facecolors, ran your recorded loop, and then it crashed....what was the "Faulting module" and "Faulting module path"? You can get these in the same place you copy/pasted in your original bug report.

aartcore, 2017-06-22:
I will test asap with a new recorded loop, i hope i will able to to do it next week.

here I've got a crashreport without the facecolors on, i had the ntdll.dll crash also a couple of times earlier with trying to isolate the bug i believe:

Faulting application name: Max.exe, version: 7.3.3.36780, time stamp: 0x58b6c502
Faulting module name: ntdll.dll, version: 10.0.10586.672, time stamp: 0x580ee321
Exception code: 0xc0000374
Fault offset: 0x00000000000ee6fc
Faulting process id: 0x14c8
Faulting application start time: 0x01d2e997e56d7ac2
Faulting application path: C:\Program Files\Cycling '74\Max 7\Max.exe
Faulting module path: C:\Windows\SYSTEM32\ntdll.dll
Report Id: 4bfa9140-3415-431d-b425-fffc90a28989
Faulting package full name: 
Faulting package-relative application ID:

diablodale, 2017-07-17:
Hello. I'm going to close this issue for now as a Microsoft bug. In your examples, it is following the pattern of the known Microsoft bug(s).
In the future, if you find another scenario when 2 dp.kinect2 objects + face causes a crash, please do add to this issue. Otherwise, please open a new issue to keep our scenarios organized. :-)

closed 2017-07-17 by diablodale