diablodale/dp.kinect2 #40: dp.kinect not registered after writing registration name and rpeg file is opened

ondeorange, 2017-07-26:
Hi Dale, dp.kinect is still not registered after writing registration name "Edgar Hemery" in the register tab and opening the .dpreg i've purchased. I believe I've correctly followed every steps (including 4, 5 and 6). I've also tried opening the trial version with the trial dpreg and I had no results either. Please help 

So we have:
Bootcamp on a MacBook Pro 2014 with Windows 10 Home version 1606, 64-bit.
Interl Core(TM) i7-3557U CPU @ 3.10 GHz
RAM 16 GB

Version Kinect v2
SDK Kinect for Windows v2.0
Kinect runtime-v2.0
VS Redistributable 2015 x64
Max 7 - registrered

diablodale, 2017-07-26:
I still need the information requested. Please https://github.com/diablodale/dp.kinect2/issues/1

ondeorange, 2017-07-26:
Setup:
Bootcamp on a MacBook Pro 2014 with Windows 10 Home version 1606, 64-bit.
Interl Core(TM) i7-3557U CPU @ 3.10 GHz
RAM 16 GB
Version Kinect v2
SDK Kinect for Windows v2.0
Kinect runtime-v2.0
VS Redistributable 2015 x64
Max 7 - registrered

Steps:
[image]

Plugged the Kinect2
Purchased the dp.kinect2 on a Mac and then transferred to an other Mac which has bootcamp without unzipping it. 
Setup guide steps 1 to 5 verified
Download dp.kinect2 
Unzip the folder, placed it in Max7/Packages folder and open dp.kinect2 help patch
Opened the dp.kinect2 help patch 
Clicked to start on the toggle 
then went to the register tab, then clicked on register with my registration name + purchased .dpreg (897) and then registration trial + trial.dpreg
Max is not responding for 20 seconds each time I click "register"

As a result, a yellow message says 'NOT REGISTRED' on the patch and the console window says "dp.kinect2 not registred or has expired. Register your key or visite ..."

    Detail every step needed to create your issue.
    If your issue was discovered in Max with a patch, please save a small patch that demonstrates the problem. Save it on the internet and provide a link to it.
    List every patch opened, button clicked, menu chosen, option selected, etc.
    Turn on the attribute @verbose=1 to receive additional debug information.
    Detail what was the bad result and what is the expected good behavior
    In the Max console window, copy the first group of messages from dp.kinect2 which provide version information and settings. Also include any later diagnostic/error messages you find; they will be in yellow or red color.

diablodale, 2017-07-26:
Thank you for the info. I have never experienced this issue with my customers before. My registration service is running with no issues and I was again successful registering with your license using two of my computers.

I do agree there is some connectivity problem between your computer and the registration service. The problem can been seen in your screenshot "WinHttpSendRequest...The operation timed out".

I looked at my code. It suggests that your computer can do a DNS lookup to license.hidale.com. However, it also suggests that it can not open a secure HTTPS connection to that same server.

I recommend that you troubleshoot connectivity issues on that computer, issues with any proxies between your computer and the internet, etc. For example, you could start by opening a command prompt and then `ping -n 4 license.hidale.com`.  You should get four perfect replies, no failures, no timeouts.

Then open your browser on the same computer. Visit `https://license.hidale.com/test/health/`
You should have no errors, the connection icon should be green (secure), and you should see the following text in your browser
```
{
  "Message": "Health is good"
}
```

If you have failures with the two above test, please contact your network administrator or Microsoft to troubleshoot connectivity problems. Those problems are outside my software and lie in your OS, hardware, or Internet connectivity.

If you believe the problem is still with dp.kinect2, please reply with the results of your connectivity tests.

diablodale, 2017-07-28:
Hello. Checking back with you. Did you resolve your issue?

diablodale, 2017-08-09:
closing this issue, customer disappeared

closed 2017-08-09 by diablodale