diablodale/dp.kinect2 #41: Issue registering DPK2 / unzipping reg file

Unicorpse, 2018-01-08:
Hi, I'm trying to register my DP Kinect 2 license (purchased June 2017!) but I can't seem to decompress the reg file to complete registration. I get the error "cannot open file as archive" with 7zip, and I can't redownload the dpreg file as I get "invalid link". 

I have Max 7 (64bit) installed and registered, working on Windows 10.

Thanks!

Unicorpse, 2018-01-08:
I also can't register a trial license - nothing happens when clicking on "register trial" (including no entries in the Max console)

Unicorpse, 2018-01-08:
Further update - after resetting Max I can now use the register buttons, and I have the DPREG file, but neither the trial or my licensed DPREG file are working for registration. Here's a screenshot of the console:

[image]

diablodale, 2018-01-08:
I see your order from June 2017. That order is for dp.kinect.
Your screenshot above shows dp.kinect2. The gen-1 and gen-2 Kinect sensors are distinct; as are the software and licenses for each.

You can download the current version of dp.kinect from https://hidale.com/shop/dp-kinect/
I have also re-enabled your license download link. You can try it again now.

With both those downloads, please follow the setup guide here https://github.com/diablodale/dp.kinect/wiki

Does that resolve your issue?

Unicorpse, 2018-01-08:
Hi Dale, thanks for your answer and good eye! That is really frustrating, I meant way back to purchase DPK2... Hence why I've made that silly mistake. 

Is it at all possible to switch over my license for DPK1 to DPK2? No worries if not.

Thanks for your help!

diablodale, 2018-01-08:
Yes, I can as you have not yet downloaded the license. 
I can make a new "exchange" order for dp.kinect2.

Can I use the same email address and license info from your original order?

Unicorpse, 2018-01-08:
That's such a big help, thank you!!!

Yes same info should all apply.

I really appreciate it! Thanks!

diablodale, 2018-01-08:
On its way via email.
I'll close this issue as resolved. If you have any new related issues with dp.kinect2, you can reopen this issue.

closed 2018-01-08 by diablodale