diablodale/dp.kinect2 #44: dp.kinect2 First Time Registration Failed

JeenaYin, 2018-07-23:
Setup: 

MacBook Pro (Retina, 15-inch, Mid 2015)
Processor: 2.2 GHz Intel Core i7
Memory: 16 GB 1600 MHz DDR3
Graphics: Intel Iris Pro 1536 MB
Bootcamp Windows 10 x64 with all current updates

Max version 7.3.5 registered 
Driver installed: Kinect for Windows Runtime 2.0
dp.kinect2-v1.1.20170110

Steps to reproduced issue:
1. First time purchased the key (Have used trial version on this laptop successfully)
2. Downloaded the dpreg file and put it in the library, and copied pasted registration name [Jeena Yin] without quotes in the text block, and clicked register and searched for the "dp.kinect-1085.dpreg" file.
3. I have never purchased any dp license or registered this one on any other laptop. I tried it multiple times and failed.
4. Max console shows: 
newobj: vidcycle: No such object
newobj: vidcycle: No such object
newobj: vidcycle: No such object
dp.kinect2 v1.1.20170110.0 Win64, Copyright (c) 2014-2017 Dale Phurrough. This software comes with ABSOLUTELY NO WARRANTY.
-All rights reserved except for those documented in the dp.kinect2 terms at http://hidale.com and granted by purchase/agreement to any entity listed as registered below.
-Compiled with Eigen 3.2.8 licensed under Mozilla Public License, v2.0. Source code & license at http://eigen.tuxfamily.org/
-BY USING THIS SOFTWARE, YOU ACCEPT THESE TERMS. IF YOU DO NOT ACCEPT THEM, DO NOT USE THIS SOFTWARE.
dp.kinect2 is not registered or has expired. Register your key or visit http://hidale.com for purchase and support.
kinect performance caution: power plan should be a high performance plan
kinect performance caution: power setting (USB selective suspend) should be disabled
kinect performance caution: power setting (Link State Power Management) should be off
kinect performance caution: power setting (Minimum processor state) should be 100%
kinect performance caution: power setting (Intel Graphics Power Plan) should be maximum performance
dp.kinect2: dp.kinect2 is not registered or has expired. Register your key or visit http://hidale.com for purchase and support.
dp.kinect2: General access denied error. hr=0x80070005
dp.kinect2: dp.kinect2 requires Kinect20.Face.x64.dll 2.0.1410.19000 or greater + the matching NuiDatabase folder from that release
dp.kinect2: RG: 167772160 RV: 0
dp.kinect2: dp.kinect2 is not registered or has expired. Register your key or visit http://hidale.com for purchase and support.

Any help would be greatly appreciated. Need to solve before presenting my work at a conference in a week.

Thanks,
Jeena

diablodale, 2018-07-24:
Hello. I see two things that could contribute to your issue. Thank you for providing good initial diagnostics

1. Above I see Max messages which indicate you are trying to use dp.kinect2 (the external made for the v2 sensor). The license you purchased is for dp.kinect (the external made for the v1 sensor). You'll need to download dp.kinect from https://hidale.com/shop/dp-kinect/ and follow the setup guide for it at https://github.com/diablodale/dp.kinect/wiki#setup-guide
2. I see errors `newobj: vidcycle: No such object`. I'm unfamiliar with that object `vidcycle`. I can't even find it in the Max forums or google search. I recommend checking/reinstalling/repairing Max and then follow the setup guide for dp.kinect again. It should be no harm to do the repeat installs.

Give those two a go and share with me the results

JeenaYin, 2018-07-24:
Thanks for the quick response! Wow that's totally my bad... Is it possible to transfer the purchase to kinect2? It's cool if it's not possible... I understand. But I only have kinect v2. :)

diablodale, 2018-07-24:
After downloading the license, I don't have a process to deactivate the v1 dp.kinect licenses. It was built with an earlier version of the licensing tech than dp.kinect2 in which there is more flexibility.

If you chose to continue with dp.kinect2, please keep me in the loop if you have any issues so we can resolve them together.

JeenaYin, 2018-07-25:
I understand. I repurchased dp.kinect2 license and it works perfectly! But I do need another dp.kienct2 license for the other kinect2... and I would not ever need the v1 dp.kinect license. 

Thanks!

diablodale, 2018-08-11:
closing this issue as resolved. If you need any other assistance related to this issue, please make another comment below.

closed 2018-08-11 by diablodale