diablodale/dp.kinect2 #45: unregister dp.kinect2 problems

circasu, 2018-08-13:
Setup
computer i7 6700 cpu , 250 gb ssd 32 gb ram ddr4 2400
Windows 10 pro 
Installed Kinect 2.0 Public Preview SDK
Installed Max 7.3.5  x64 and it is registered
Using dp.kinect2 version 1.1

I want to remove the registry of this computer to use it in a new computer, but although it confirms to me that the registry has been removed it does not arrive it to realize and I can not authorize the new computer.
The features of the new computer are:

zotac model in 1060k, cpu i5 7500, ram 16 gb ddr4 2400, ssd 250 gb. Gpu zotac nvidia 1060.

version of max msp 7.3.5 x64 authorized.

using dp.kinect2 version 1.1

[image]
[image]
[image]

diablodale, 2018-08-14:
I may have reproduced something on my Windows 10 machine. I need to debug this further.

To get you running now, I might have a workaround. Please try the following on all three of your machines:

1. In Windows, open a command prompt
2. Type `md "%localappdata%\Dale Phurrough"`
3. press enter
4. Type `md "%localappdata%\Dale Phurrough\dp.kinect2"`
5. press enter

Now on your two old machines, do the process I previously described:

1. Unregister
2. Register
3. Unregister

Then on your new machine:

1. Register

circasu, 2018-08-15:
Hello, with the instructions that you have given me, I have been able to register on the new computer.
From what I understood, this is just a temporary fix, is not it? I beg you that when you solve this issue you get in touch with me so that you can make the records in a normal way.

Thanks for everything and greetings.

diablodale, 2018-08-15:
Good to hear that resolved your issue. Making those two directories fixes the problem on those three machines. You don't have to do it again. I have added this workaround in the setup guide for dp.kinect2 so that other people won't have this problem while I work on a code fix.

Thank you for your patience and diagnostic information so that I could isolate the issue.

diablodale, 2020-06-29:
I've released the v1.2 feature update to resolve this issue directly in the code. Available to download at the usual locations like https://hidale.com/shop/dp-kinect2/

closed 2020-06-29 by diablodale