diablodale/dp.kinect2 #46: dp.kinect2 hardware unavailable for request after a few seconds

AlBaum, 2019-01-21:
Hi,
As the title says, when I start dp.kinect2 (this also happens in the help file) after a few seconds of seeing the image, it freezes and I get the message dp.kinect2 hardware unavailable for request in the max window. 
I thought it might be due to a problem with the USB adaptor I am using -I am on Bootcamp on a macbook pro with only USB-c ports and need adapter - BUT: the strange thing is that if I open the Kinect Studio (from SDK) at the same time as dp.kinect, and activate an image in the kinect studio, dp.kinect2 runs fine and does not show any error.
Any idea what could be the problem ?

I am on Windows 10 Home on Bootcamp
Macbook pro 2017 (touchbar): Intel Core i7 6700HQ @2.6ghz
Max 7.3.5
Tried with three different USB-c ->USB3 adapters

Thanks in advance for any hint,
Alain

diablodale, 2019-01-21:
The only idea that I have is a Kinect driver incompatibility with the Mac and/or the adapter and/or the Intel processor. I have no way to test this as I don't have a Mac and I am not aware that Microsoft supports the Kinect on a Macintosh. Your USB port/adapter must support the full USB3 featureset, which includes at a minimum a Gen2 PCIe port, full isochronous support, etc.
I recommend you use the Kinect Configuration Verifier. You can get to it from step 1 of my setup guide at https://github.com/diablodale/dp.kinect2/wiki#setup-guide

You typed dp.kinect once above. But I'm assuming you mean dp.kinect2. Please correctly if otherwise.

There are known issues with Windows, the Kinect drivers, and some of the recently Intel processors as it related to power management. This could be your issue.  Here are some links to Microsoft documentation on these:
https://social.msdn.microsoft.com/Forums/en-US/7a9f139d-85fe-4e68-9ad2-5456f8e1f173/kinect-v2-disconnects-and-reconnects-on-recent-intel-chips-solution?forum=kinectv2sdk

And Microsoft isolated and fixed a bug recently with Windows 10 version 1809, you can read about that here:
https://social.msdn.microsoft.com/Forums/en-US/a67df395-0365-4108-9254-2f25f354795f/kinect-for-windows-runtime-update-to-address-issues-on-windows-10-1809?forum=kinectv2sdk

Also be sure to address any warnings you get in the Max console. Sometimes dp.kinect2 will give you suggestions on potential problems.

If none of the above helps, please read https://github.com/diablodale/dp.kinect2/issues/1 and attach all the information described. I might see something in that data that could assist.

diablodale, 2019-01-21:
Also, the Kinect Studio for v2 has a low level connectivity directly to the Kinect driver in Windows. You write above "...activate an image...".  What does that mean?

AlBaum, 2019-01-22:
Thank you so much dablodale for your prompt and exhaustive reply. It works now.
I managed to solve the problem with the solution offered in the first link you gave:

https://social.msdn.microsoft.com/Forums/en-US/7a9f139d-85fe-4e68-9ad2-5456f8e1f173/kinect-v2-disconnects-and-reconnects-on-recent-intel-chips-solution?forum=kinectv2sdk
Solution is :
From an elevated cmd prompt, type:
reg add HKLM\System\CurrentControlSet\Control\Processor /v Capabilities /t REG_DWORD /d 0x7e060

And for information in case anyone else comes across the problem here is the full description of setup and problem:
Computer: MacbookPro with Touchbar (2017)
OS: Windows 10 Home 1809 running on Bootcamp
Kinect driver version: 2.2.1811.10000
Kinect is connected to Computer via a Usb-c to USB3 converter
Max version 7.5.3 (demo mode)
dp.kinect2 version 1.1.20170110

The symptom was that dp.kinect2 would work for a few seconds at a time and stop working (light went off on Kinevt) withthe message "dp.kinect2 hardware unavailable for request" appering in the max window. If connecting the Kinect at the same time in the MIcrosoft Kinect Studio, the problem would disappear and dp.kinect2 worked. Solved the problem as above.