diablodale/dp.kinect2 #49: Kinect Studio Record Doesn't work anymore

silicatproject, 2019-01-30:
Hi,
It's not a dp.kinect2 issue but since i made the windows update i think but not 100%sure , with kinect studio i'm unable to play a recorded kinect sequence ?
Really strange , it worked some days ago ...
Have you the same problem on your side  ?
thanks

	"version" : "Max Version 8.0.4 (f58f8b9) (64-bit windows)",
	"platform" : "windows",
	"arch" : "x64",
	"osversion" : "Microsoft Windows 10 Professional (build 17763), 64-bit"

Win10 Pro
version 1809
installed 1/29/2019
osbuild 17763.292

KinectRuntime-v2.0_1409
KinectSDK-v2.0_1409

dp.kinect2 v1.1.20170110.0 Win64

diablodale, 2019-01-30:
Hi. I don't support Kinect Studio or the Kinect drivers themselves. You will need to seek that support from Microsoft.

Personally, I haven't used Kinect Studio in years. I have no knowledge of its current functionality.
dp.kinect2 and the Kinect drivers on my computers continue to work with no problems.

Microsoft has acknowledged problems with their Kinect drivers/software and has so far provided two updates/workarounds for them. Perhaps these Microsoft issues are what you are encountering. You can find those workarounds/updates on the official Microsoft Kinect forum and in already closed issues already posted here.
https://github.com/diablodale/dp.kinect2/issues/46
https://github.com/diablodale/dp.kinect2/issues/47
https://github.com/diablodale/dp.kinect2/issues/48

silicatproject, 2019-01-30:
Thanks you so much

diablodale, 2019-02-02:
Hi. Checking back with you. As a potential help in the future (collective knowledge), did any of those Microsoft fixes/workarounds resolve your issue with Kinect Studio?

diablodale, 2019-02-05:
I'm closing this issue as it is not related to dp.kinect2.
Keeping the conversation as it might help someone if they search and find this. :-)

closed 2019-02-05 by diablodale