diablodale/dp.kinect2 #50: Cant fetch reg file from disk (Automated registration issue)

klippancollective, 2019-02-04:
Hello Dale!

My issue is that i cant seem to automate the register message (register examplename c/pathway/file.dpreg) to the dp.kinect2. It is possible to do so with the trail message, but Max is unable to find another file other than the trial.dpreg. 

Result:
The error message i get is "registrationfile not found"

Dell XPS 15 9570 Laptop, 32gb ram, 1tb ssd, intel core 17 8750H
Windows 10, 64 bit
Cycling 74's Max 8.03
KinectSDK-v2.0_1409 
Dp.kinect2 v. 1.1

**Steps**

Download this patcher https://we.tl/t-lKM5GTvoTV
Open it and build max application
Put the reg file in the same folder
Press on message. (there is a hidden dp.kinect2 object beneath the message, visible in edit mode)

I will send reg file with email. 
Kind regards

Anton

diablodale, 2019-02-04:
Hello. I don't yet have enough information to setup the repo case. Please add more detail.
I need to know every click and button and menu you press/choose/select.
I need to know every directory name and patch that is involved. For example:

- Where is Max installed?
- How did you install dp.kinect2? Where is it installed?
- Where do I save the downloaded patch?
- What does "build max application" mean?
- If I create an MXF or EXE, where do I save it?
- etc.

klippancollective, 2019-02-04:
Max is installed in "C:\Program Files\Cycling '74\Max 8\"

 

Dp.kinect2 is installed locally on my computer in the package folder in Max 8\resource\packages

I am assume that you already have max installed on your machine, so it is not included in the file I sent you since it is only a single patch.

I build the application by first making the file into a project file and then adding dp.kinect into the project before building it. Then build to exe.

When building application, the object “sendapppath” will automaticly locate the folder that the exe or mxf is stored in. Save in any folder.

 

 

 

 

I must ask you, Do you have any examples you can show me on your take on automated register patches? 

If so, please send me an example so I can study it.

diablodale, 2019-02-04:
I still need every directory I requested. You gave me Max and dp.kinect2. I now need all the others.
I asked for these details because I have to setup the EXACT same hard drive and paths that you have on your machine. If I don't, then I can not adequately reproduce your problem.

You write...

> I build the application by first making the file into a project file and then adding dp.kinect into the project before building it. Then build to exe.

Please detail the process with every click and button you press. Max should automatically include all externals that are used. Meaning, you should not have to use some UI to manually select the dp.kinect2.mxe64 file. It is these unusual steps that lead to me asking for every detail of what you are doing.

The example of automating registration is in the help file included with dp.kinect2. On the registration tab. That tab uses it every time a customer registers dp.kinect2. You can see it with the trial and with a paid registration file.

diablodale, 2019-02-04:
While you compile that information, I have found two issues so far
Max is quite poor at passing messages with strings that have spaces in them. Max very quickly changes them into separate values on the same message.

1. You have to put double quotes around your registration name. Without using the double quotes, you are not sending:
(register "Firstname Lastname")
instead you are currently sending
(register "Firstname" "Lastname")

2. The same issue occurs with the path. Max is changing your path into separate values. This is not easy or obvious, Max is working against us both.

Using your example patch, put a (zl len) to receive the message you composed.
zl should report 3
Instead your patch reports 7 because Max has seen the spaces and converted them all into separate values.

diablodale, 2019-02-04:
I have your patch working after some fixes. I have not tested this inside a built app.

The two fixes for the two issues I wrote above are:
1. Put the quotes around your registration name:   (prepend register "my registration name")
2. Put symout on your sprintf so that the string created is sent as a single string: (sprintf symout %s%s)

With these two changes, your patch found the registration file and attempted to register.

klippancollective, 2019-02-06:
Thanks a bunch, dale!

 

Real solid help. It worked out with putting quotes in the register message and “symout” in the sprintf object. 

Finally got it working, no issues. Thanks!

Wonderful support!

 

/Anton

diablodale, 2019-02-06:
great to hear. I'll close this issue as successfully resolved.
I will also update the wiki based on this issue.

closed 2019-02-06 by diablodale