diablodale/dp.kinect2 #51: Cannot register DPK2

Harveynglish, 2019-05-29:
Hello.

I had a trial version of DPK2 and it worked fine. When I purchased the full version, I have been unable to register my key at all. 

I have uninstalled both max 8 and DPK2 and still having the same issue.

I open the DPK2 help file.
Go to register tab.
Enter name: Harvey English
Click Register.
direct to the unzipped DPREG file
Then receive this message:

dp.kinect2: RG: 167772160 RV: 0
dp.kinect2: dp.kinect2 is not registered or has expired. Register your key or visit http://hidale.com for purchase and support.

System is Lenovo IdeaPad running windows 10 home 64bit - version 1803 Build 17134.706

Please help, this is the last hurdle in a project

diablodale, 2019-05-29:
According to my records, you purchased a license for dp.kinect.
Above you pasted text for dp.kinect2
Those are two different sensors, different software, different licenses.
Is there another license you are using?

Harveynglish, 2019-05-29:
No sir, but that makes alot of sense. If that is the case then it seems that I have purchased the incorrect product. Would it be possible for you to change my purchase to the DPK2? I have no use for the dp.kinect license.

Sorry for the confusion

Harveynglish, 2019-05-30:
No sir, I do not have an other licenses, but that makes sense. I must have
clicked on the wrong purchase. If that is the case then it seems that I
have purchased the incorrect product. Would it be possible for you to
change my purchase to the DPK2? I have no use for the dp.kinect license.

Sorry for the confusion

*Harvey English*

*PH: 512-529-8866| Maker | Creator | Doer | Mentor@ TEDxYouth@Austin
<http://tedxyouthaustin.com/> |  LinkedIn
<http://www.linkedin.com/pub/harvey-english/65/995/851/> |  harveymakes.com
<http://harveymakes.com>*

diablodale, 2019-05-30:
The dp.kinect licensing technology is not able to switch a license to other software after you have downloaded the license. I treat all my customers equally and fairly with returns + the tech just can't do it.
The newer dp.kinect2 does have more advanced tech; I could have done that in reverse.

While this may not be the answer you prefer, I hope you understand the logic of how I got there.
I do appreciate you considering dp.kinect(2) for your solutions. If you want to purchase a license for dp.kinect2 (the 2nd sensor - the one with XBox One, larger, and has hard rectangular corners) you can purchase it at the same place you downloaded the trial https://hidale.com/shop/dp-kinect2/

Harveynglish, 2019-05-31:
Dale, I understand your policy. But let me explain my situation.

I have the newer sensor. I have not ever owner the first generation. I was
unaware that there was different software for the different sensors. I have
spent well over 1000$ on a project using this kinect to control fire
effects giving the user the feeling of controlling fire with their
movement. This was a personal project pursued purely for fun and making
cool art. As a result I am financially broke and this is the last piece of
the puzzle.

I understand that you are unable to switch the Licence, but I do know that
you have the ability to deactivate or revoke a license. Wouldnt it then be
possible to deactivate my dp,kinect license and reissue me a dp.kinect2
license?

*Harvey English*

*PH: 512-529-8866| Maker | Creator | Doer | Mentor@ TEDxYouth@Austin
<http://tedxyouthaustin.com/> |  LinkedIn
<http://www.linkedin.com/pub/harvey-english/65/995/851/> |  harveymakes.com
<http://harveymakes.com>*

diablodale, 2019-06-03:
resolved this offline in email :-)

closed 2019-06-03 by diablodale