diablodale/dp.kinect2 #53: Register key doesn't work on same computer

edavidova, 2019-10-21:
Hi, I had to wipe out my PC and now can't register the dp kinect. Please let me know if there is any solution, thank you!
Eva

diablodale, 2019-10-21:
Hello. There are no known issues with registration. I have no information on which I can help you. 

You can review all the setup and registration steps at https://github.com/diablodale/dp.kinect2/wiki to learn how to install it again on your Windows computer.

Past resolved issues from other customers are also available here by clicking "issues" above, and then "closed" issues. If none of those helps you, please read https://github.com/diablodale/dp.kinect2/issues/1 as it has several questions and information which I need to help you further.

I'm confident we can resolve this quickly after we share the needed information.

edavidova, 2019-10-22:
Thank you Dale,
It just says that my keys have expired.

diablodale, 2019-10-22:
please read #1 as it has several questions and information which I need to help you further.

edavidova, 2019-10-23:
Thank you,
I will send everything tomorrow

edavidova, 2019-10-23:
Hi Dale,
Here are the specs for my computer and setup. It ran for a year, but I had
to wipe it and reinstall everything. The only difference is that I
installed Max 8 instead of 7.

Intel(R) Core(TM)i7-7700HQ CPU @2.80GH 2.80GHz
RAM 16 GB
Device ID C6377767-FB54-4046-824E-37C7B23FB8AC
System 64-bit operating system, x64-based processor

Windows 10 Home
version 1903
OS build 18362.356

Kinect SDK v2.0 1409

*Thank you again!* Looking forward to the solution!

All the best,
Eva
www.evadavidova.com

diablodale, 2019-10-23:
I am willing to help. However, I first need answers to all the questions in https://github.com/diablodale/dp.kinect2/issues/1
So far, you have only answer 2 of the 10 questions there.

edavidova, 2019-10-24:
Thank you Dale, and forgive my misunderstanding, and my little experience.

A friend (great musician and artist Matthew Gantt) helped me put the Max
project together, so, although I can see what's going on, I am not an
expert at all.
I do realize now that probably I should have *Unregistered*, and then *Register
*again after the wipe up of the computer...Too late. I would buy another
licence, but the computer might not perform after all, so I don't want to
"waste" a new licence before I know if the PC will behave.
__________
1 Setup:
Intel(R) Core(TM)i7-7700HQ CPU @2.80GH 2.80GHz
RAM 16 GB
Device ID C6377767-FB54-4046-824E-37C7B23FB8AC
System 64-bit operating system, x64-based processor
Nvidia Geforce GTX 1070

Windows 10 Home
version 1903
OS build 18362.356

Kinect SDK v2.0 1409

Max 8 registered (i used Max 7 before, same registration)

2. Issue:
(Registration not working. Console messages bellow:)
dp.kinect2 is not registered or has expired. Register your key or visit
http://hidale.com for purchase and support.
dp.kinect2: dp.kinect2 is not registered or has expired. Register your key
or visit http://hidale.com for purchase and support.
dp.kinect2: General access denied error. hr=0x80070005

dp.kinect2: dp.kinect2 requires Kinect20.Face.x64.dll 2.0.1410.19000 or
greater + the matching NuiDatabase folder from that release* (I could not
find this on the internet and don't need Kinect face)*
dp.kinect2: registration key is not valid for activation on this date
2019-10-24T15:34:27Z
dp.kinect2: RG: 167772160 RV: 0
dp.kinect2: dp.kinect2 is not registered or has expired. Register your key
or visit http://hidale.com for purchase and support.

The @verbose 1 is already on the dp.kinect2 patch.

On my artwork's patch, there is one additional message:  newobj: OSC-route:
No such object
Not sure if it is related to dpKinect

*As I don't know how to put them on the internet, 2 zipped patches are
attached-- the dp.kinect2 patch, and the patch for the artwork.*
  __________

I hope this is enough information... And  hope we can make it work for
tomorrow as I have to show to some people.
*THANK YOU again!*

All the best,
Eva

diablodale, 2019-10-24:
Hi, I am not able to provide support for your application made by your friend.
I recommend you simplify first. Only follow the setup and registration steps at https://github.com/diablodale/dp.kinect2/wiki.

I am still missing a lot of information. I need answers to all ten of the questions I asked.  To remind you....
- What specific version of Max are you running?  8 dot what? 8.?.?
- What version of the dp.kinect2 external you are using.
- Steps to reproduce your issue. Detail every step needed to create your issue. Please read #1  and provide detailed steps. I have six different things there I need.
- In part of the console output you provided, I can see that something was done to dp.kinect2. Mayne you or the software tried to register twice. This is unclear to me and this is why I need to know all the steps you follow, everything, all clicks, all installs, everything. You can see an example of the detail I need in #1 
- I need *all* the max console output. Copy and paste from the top to the bottom of the Max console window. There is much Max console output that you didn't provide.

It is ok to format your hard drive and reinstall dp.kinect2 on the same computer. When you register, dp.kinect2 should see that your hardware is the same and re-connect to the same license.

OSC-route is not software I provide. I recommend you google for it to learn more.

edavidova, 2019-10-24:
Hi Dale,
The problem is the same even if I only open a new patch with dpkinect. It just doesn’t recognize the license. Could you please give me a new one or access to trial? I promise if everything works I’ll pay once more

diablodale, 2019-10-25:
Is there a language barrier between us? What are the languages you best write? I can write in English or in German.

I still can not help you until you provide answers to the questions I asked.

edavidova, 2019-10-25:
Hi Dale,
I think is more of a knowledge barrier--you know a lot, I know much less. I
will try to respond everything and hope it will work soon! I would love to
show the piece today.

   - What specific version of Max are you running? 8 dot what?* 8.1.0.59272*
   - What version of the dp.kinect2 external you are using. *Indeed there
   are 2 dp.kinects2: One is dp.kinect2.mxe64 and the other,  dp.kinect2.mxe
   . They both have the same owner: Desktop-B038LQ3\Eva*
   - Steps to reproduce your issue. Detail every step needed to create your
   issue. Please read #1
<https://github.com/diablodale/dp.kinect2/issues/1> and
   provide detailed steps. I have six different things there I need.* I
   really don't know what else to provide. I might be missing something that
   is obvious for you, but not for me...*
   - In part of the console output you provided, I can see that something
   was done to dp.kinect2. Mayne you or the software tried to register twice.
   This is unclear to me and this is why I need to know all the steps you
   follow, everything, all clicks, all installs, everything. You can see an
   example of the detail I need in #1
   <https://github.com/diablodale/dp.kinect2/issues/1> *Unfortunately I
   don't remember all that I, but it does look like two files (one mxe, and
   one mxe64)*
   - I need *all* the max console output. Copy and paste from the top to
   the bottom of the Max console window. There is much Max console output that
   you didn't provide.* Please see bellow, thank you!*

*on the main dp.kinect2 patch window:*
you have multiple files in your search path with the name dp.kinect2.maxhelp
  using this file: C:/Users/Eva/Documents/Max
8/Packages/dp.kinect2/help/dp.kinect2.maxhelp
here are the other files with this name:
  C:/Users/Eva/Desktop/dp.kinect2.maxhelp
dp.kinect2 v1.1.20170110.0 Win64, Copyright (c) 2014-2017 Dale Phurrough.
This software comes with ABSOLUTELY NO WARRANTY.
-All rights reserved except for those documented in the dp.kinect2 terms at
http://hidale.com and granted by purchase/agreement to any entity listed as
registered below.
-Compiled with Eigen 3.2.8 licensed under Mozilla Public License, v2.0.
Source code & license at http://eigen.tuxfamily.org/
-BY USING THIS SOFTWARE, YOU ACCEPT THESE TERMS. IF YOU DO NOT ACCEPT THEM,
DO NOT USE THIS SOFTWARE.
dp.kinect2 is not registered or has expired. Register your key or visit
http://hidale.com for purchase and support.
dp.kinect2: dp.kinect2 requires Kinect20.Face.x64.dll 2.0.1410.19000 or
greater + the matching NuiDatabase folder from that release
dp.kinect2: dp.kinect2 is not registered or has expired. Register your key
or visit http://hidale.com for purchase and support.
dp.kinect2: General access denied error. hr=0x80070005

It is ok to format your hard drive and reinstall dp.kinect2 on the same
computer. When you register, dp.kinect2 should see that your hardware is
the same and re-connect to the same license.

*On the registration tab:*

you have multiple files in your search path with the name dp.kinect2.maxhelp
  using this file: C:/Users/Eva/Documents/Max
8/Packages/dp.kinect2/help/dp.kinect2.maxhelp
here are the other files with this name:
  C:/Users/Eva/Desktop/dp.kinect2.maxhelp
dp.kinect2 v1.1.20170110.0 Win64, Copyright (c) 2014-2017 Dale Phurrough.
This software comes with ABSOLUTELY NO WARRANTY.
-All rights reserved except for those documented in the dp.kinect2 terms at
http://hidale.com and granted by purchase/agreement to any entity listed as
registered below.
-Compiled with Eigen 3.2.8 licensed under Mozilla Public License, v2.0.
Source code & license at http://eigen.tuxfamily.org/
-BY USING THIS SOFTWARE, YOU ACCEPT THESE TERMS. IF YOU DO NOT ACCEPT THEM,
DO NOT USE THIS SOFTWARE.
dp.kinect2 is not registered or has expired. Register your key or visit
http://hidale.com for purchase and support.
dp.kinect2: dp.kinect2 requires Kinect20.Face.x64.dll 2.0.1410.19000 or
greater + the matching NuiDatabase folder from that release
dp.kinect2: dp.kinect2 is not registered or has expired. Register your key
or visit http://hidale.com for purchase and support.
dp.kinect2: General access denied error. hr=0x80070005

dp.kinect2: registration key is not valid for activation on this date
2019-10-25T15:38:55Z
dp.kinect2: RG: 167772160 RV: 0
dp.kinect2: dp.kinect2 is not registered or has expired. Register your key
or visit http://hidale.com for purchase and support.
dp.kinect2: RG: 33558528 RV: 0
dp.kinect2: dp.kinect2 is not registered or has expired. Register your key
or visit http://hidale.com for purchase and support.
*I fervently hope this will help resolve the problem. Thank you for your
time, and please forgive my inexperience.*
*All the best,*
*Eva*

*PS: **Today is my birthday in case this helps.*
*PS.PS <http://PS.PS>: One of the good things about this back and forth has
been to see some of your work. And I think I've seen "Birds in a Tree" way
back,  years ago... Are you going to do any show/ festival in NY soon? *

diablodale, 2019-10-25:
Also, are you clicking on the (trial) button on the registration tab?
I think you already own a dp.kinect2 license; I see a similar named person in my store's records. Paid licenses have no expiration date. So one of the errors you see `registration key is not valid for activation on this date` is not possible with a paid license.

edavidova, 2019-10-25:
I am going to delete the dp.kinect and download again to see if anything
works better.
Please let me know where should I put the *.dpreg file*... Or if there is
anything different that I should do
*Thanks!*

....registration key is not valid for activation on this date is not
possible with a paid license." --It is possible, because it is what is
happening in my case :-(

edavidova, 2019-10-25:
Hi Dale, I deleted and re-downloaded the dp.kinect2 folder. Still the
same... Is it possible that the .dpreg file doesn't work because it was on
an external drive? (while I had all the PC reformatted)
I am really dishearted right now.

diablodale, 2019-10-25:
I can help you no further until you answer my questions I asked above. From what I can see so far, everything is working as expected.

diablodale, 2019-11-04:
Checking back in with you. Did you resolve the issue yourself? If not, I can help after you provide the diagnostic information I requested above.

If I don't hear from you in roughly a week, I'll close this issue.

diablodale, 2019-11-09:
Closing due to no information from customer

closed 2019-11-09 by diablodale