diablodale/dp.kinect2 #54: Registration doesn't work

OttavioFavet, 2019-12-17:
## Description

Please provide a clear and concise description of your issue.

## Setup

* Description of your computer hardware
- intel i5-4570s CPU @2.90GHz
- RAM 8 gb
* Version of your Operating System and any patches/updates
- Windows 10 education 64 bit
* Version of Microsoft's Kinect v2 drivers (or SDK) you have installed
- v2.0_1409
* Version of Cycling74 Max you have installed and if it is registered
-Max 8.1.1 registered
* Version of the dp.kinect2 external you are using
- dp.kinect2-v1.1.20170110
Hi Dale,
I'm not able to register my product, can you please help me?

Error: 
[image]

diablodale, 2019-12-17:
Hello. I saw in my store you recently purchased a license. Looking then at your screenshot, I see that you are not using the registration name that was sent to you by email with your order. Please check your email. You probably received 3 emails from my store. One of those emails has your registration name. You must use that registration name. I suspect this is the issue.
https://github.com/diablodale/dp.kinect2/wiki#register-dpkinect2

If you continue to have difficult registering, the error code in your screenshot could also appear if your dpreg license file is corrupt. I can easily enable your download again if we need to.

OttavioFavet, 2019-12-17:
Hello,
thank you for the quick response. I didn't noticed the capital letter on registration name. It's working now, thank you very much.

diablodale, 2019-12-17:
Glad to hear it is working for you. :-)
Happy patching!

closed 2019-12-17 by diablodale