diablodale/dp.kinect2 #55: Unable to start speech recognition

gavspav, 2020-03-09:
Setup:

Dell 9560 Laptop  I7 7700HQ, 16GB Ram, 500GB SSD
Windows 10 Home 10.0.18363
Kinect Drivers: 2.2.1905.16000
Max 7.3.4 64bit Registered (have tried with 32bit and Max 8 unregistered)
DPKinect2v1.120170110.0 Win 64

I did have stop a bunch of services recently to get the laptop to perform satisfactorily with audio. I thought I had restarted them all but perhaps I have missed one.

## Steps to reproduce

1. Start Max 7 64 bit.
2. Open the dp.kinect2 help patch
3. Click on the soundspeech tab
4. Click the checkbox labeled "start"

## Result

[image]

## Expected Result

Speech Recognition!

## Max console output

```
With verbose 1 just get an additional message Kinect has not not been successfully opened and initialised.
```

diablodale, 2020-03-09:
Hello. There are no known issues with speech on dp.kinect2 and Windows 10.

1. I don't see you followed the setup guide. Please visit https://github.com/diablodale/dp.kinect2/wiki and follow each step. For example, I see nothing about how you installed the speech sdk, language packs, etc.
2. I need all the Max console output.
3. Can you run the demo on the 1st tab 'dp.kinect2'? You should see a depthmap and color image.
4. Please use the Microsoft Kinect Configuration Verifier. You can download it from https://developer.microsoft.com/en-us/windows/kinect/ and ensure you get green checkboxes for all the tests.
5. I do support dp.kinect2, but I can not provide help with your Windows audio problems. Microsoft provides support for their OS, audio, and their speech technology. Stopping services is a very very advanced user action and extremely rare to do. It is my experience that no computer made after 2012 needs to stop services to play audio. This is very suspicious and I recommend you return your computer to its standard services configuration. And only after you successfully have audio and speech, return to tuning your computer with advanced service configuration changes.

Please review the above 5 and I will wait on your answers.

gavspav, 2020-03-09:
1. Yes I followed the guide to install the speech sdk.
2.[image]
3.Yes all the other tabs seem to work fine.
4.The USB Controller section has a warning "Your USB Configuration may support Kinect for Windows"
5.Unfortunately I did stop some services in an attempt to reduce DPC latency to  acceptable levels to use real time audio for an installation. I will check again that I have restarted all those services.

diablodale, 2020-03-09:
To further check how the speech dependencies were installed, please send a cropped screenshot of the following:
* press window-I to go to windows settings.
* click Apps, search for "speech"
* click on Kinect for windows speech recognition language pack (en-us)
* make a cross screenshot of that and its version number
* click on Microsoft server speech platform runtime (x64)
* make a cross screenshot of that and its version number
* please attach both screenshots

gavspav, 2020-03-09:
[image]
[image]

diablodale, 2020-03-09:
First problem I see is your language pack. The demo requires **en-US**. You don't have that installed. You only have en-gb installed.

More testing. I compiled the Kinect v2 SDK speech sample. This sample is 100% microsoft. Does it work for you?

 Watch the status bar in the app at the bottom. It should eventually give you instructions for words to say...and the turtle will move. If you get error messages, please tell me what it writes. It might many seconds while it starts, turns on your Kinect, loads the language, etc.

[SpeechBasics-D2D.zip](https://github.com/diablodale/dp.kinect2/files/4308609/SpeechBasics-D2D.zip)

gavspav, 2020-03-09:
Just installed the US version and now it works! Thanks. 
And yes i have a moving turtle.
Thanks so much for your help.

diablodale, 2020-03-09:
No worries. The speech technology for the v1 and v2 Kinect sensors is... mmm... specific in how it works. ;-)

On that patch tab, to the left of the yellow box, you can change `@speechlocale` to something else. And edit the `.grxml` file to match that new language + add new words. Maybe "cheers", "mate", and other GB dialect words.

gavspav, 2020-03-09:
Cheers Mate!

closed 2020-03-09 by diablodale