diablodale/dp.kinect2 #57: Not able to register purchase

krisforce, 2020-05-06:
## Description

When I go through the steps to register dp.kinect2 it fails. I get console errors that say "dp.kinect2 is not registered or has expired" , "general access denied"

## Setup

* Computer i7-7700HQ, 256GB SD, 16GB Ram
* Windows 10 Home - version 1903
* Installed Kinect for Windows SDK 2.0
* Installed MAx 8.1.3, registered
* Using dp.kinect2 version - dp.kinect2-v1.1.20170110
* ...

## Steps to reproduce

1. Open Max
2. create new patch
3. create new object dp.kinect2
4. right click to open dp.kinect2 help
5. go to registration tab
6. enter registration name from email, make sure it is spelled correctly and press button
7. file browser opens, locate dp.kinect2 .dpreg file, press open
8. go to first tab in help
9. click to start

## Result

get console error "dp.kinect2 is not registered or has expired" , "general access denied"

## Expected Result

I expect the kinect to light up and work with the help patch like it did with the trial registration

## Workarounds

no workarounds. Have tried many things, such as re-installing dp.kinect2, updating Max, rebooting... just really stubbornly won't register. Please note, my machine and kinect set up are working with TouchDesigner, Ableton, Unity 3D and Brekel softwares so it's probably not my system, hardware or kinect SDK. Also, I have not used an older kinect v1 with this machine so there should be no registration conflicts. 

## Max console output
[image]

```
When you set `@verbose 1`, additional debug messages were sent to the Max console.
Please copy and paste here...
1. The first group of messages from dp.kinect2 which provide version information and settings.
2. Include any diagnostic messages you find related to your issue. They will be in yellow/red color.
```

diablodale, 2020-05-06:
Hi. The diagnostic codes that I can see are consistent with what I previously wrote. So I want to get more diagnostics to resolve this.

Would you please follow your repro steps 1-7 again and capture the Max console output?
The screenshot you attached shows a lot of activity that is hiding diagnostic information.
Please only do steps 1-7. Don't click on any part of tab 1. And please ensure that the full Max console window is in the picture. Therefore, it should show the dp.kinect2 startup information like its version, 32/64-bit, copyright, etc.

At step 7, what happens on that registration tab. Do you see a green "registered" or yellow "not registered"?

And is this machine connected to the internet? And is Max itself allowed to connect to the internet (perhaps your have a firewall or virus blocking it? This shouldn't cause this failure. However, if it is connected to the Internet, I might be able to see something in my license server logs.

diablodale, 2020-05-06:
While you gather the above three things, please look specifically at the dp.kinect2 version that appears in the Max console when it first starts up. One possibility that would match your error code, is if you are trying to use your newly bought license with an early (v0.xxx) release of dp.kinect2. You might have such a pre-release version of dp.kinect2 on your Max path. Max has a very specific order of directories it checks when you use the method in step 3 of your repro steps.

krisforce, 2020-05-07:
Hello, yes, I am connected to the internet. I repeated steps 1-7 and see the error code in the attachment. 

There was a lot of activity in the previous capture because I added @verbose 1 to tab one in help and continued to step 8. 

[image]

diablodale, 2020-05-07:
Hi. Unfortunately, that 2nd screenshot doesn't match my needs.

Would you please follow your repro steps 1-7 again and capture the Max console output?
The screenshot you attached shows a lot of activity that is hiding diagnostic information.
Please only do steps 1-7. Don't click on any part of tab 1. And please ensure that the full Max console window is in the picture. Therefore, it should show the dp.kinect2 startup information like its version, 32/64-bit, copyright, etc.

krisforce, 2020-05-07:
Hello Dale, I was not able to see more data from the console because the kinect was not initializing. I did find a earlier instance of the dp.kinect2 external nested inside a machine learning demo patch. Once I deleted this patch, I was able to register without incident. My issue is now resolved. Kind regards.

diablodale, 2020-05-07:
Good to hear. That was the problem. Max was searching its path and found that old v0.x copy and loaded it before the v1.x version.

If you want, you should be able to run that machine learning demo also. Remove the old mxe/mxe64 within it. Then Max will find your newer mxe/mxe64 when it loads that ML patch.

Your license is good for any v1.x version. However, it doesn't work retroactive for old v0.x you might find floating around the Internet. I *do* grandfather old licenses forward. But I don't time-travel backwards. ;-)

Happy patching! :-)

closed 2020-05-07 by diablodale