diablodale/dp.kinect2 #58: Registration Issue

jdjddjdj, 2021-09-13:
## Description
Hello Dale!
Amazing software - thank you

The problem:
I purchased DP.kinect2 tonight.
I get these errors in the console after entering my user name ( John Daniel ) >  clicking "Register" >  selecting using the dialog the file dp.kinect2.dpreg

dp.kinect2: RG: 33556480 RV: 0
dp.kinect2: dp.kinect2 is not registered or has expired. Register your key or visit https://hidale.com for purchase and support.

FYI - the Trial version of DP.Kinect2 ran perfectly

*****     MY THINKING AS TO CAUSE OF ISSUE    *******
After much looking at the dp kinect forums - I believe I may have a corrupted dpreg file...  the internet here is extremely unreliable and drops out constantly, even very small downloads can be problematic.    Could this cause this error?    

## Setup

* Description of your computer hardware
* Version of your Operating System and any patches/updates
* Version of Microsoft's Kinect v2 drivers (or SDK) you have installed
* Version of Cycling74 Max you have installed and if it is registered
* Version of the dp.kinect2 external you are using
* ...

Alienware X51_R2
4gb ram
64 bit os x 64 based proc
Windows 10 pro    Vers 20H2      OS build 19042.631
Windows Feature Experience Pack 120.2212.31.0
Latest Kinect drivers as of one month ago (how do i retrieve this info?)

MAX support info   (unregistered Max currently)
Max V 8.1.11 (17de540) 64 bit windows

{
	"version" : "Version 8.1.11 (17de540) (64-bit windows)",
	"platform" : "windows",
	"arch" : "x64",
	"osversion" : "Microsoft Windows 10 Professional (build 19041), 64-bit",
	"samplerate" : 44100,
	"iovs" : 512,
	"sigvs" : 64,
	"scheduler_in_audio_interrupt" : "off",
	"audio_drivername" : "ad_mme",
	"audio_driver_subname" : "",
	"license" : "demo",
	"machine_id" : "42f9db7dbafc50d652f7e898bf215699",
	"eventinterval" : 2,
	"schedinterval" : 1.0,
	"overdrive" : "off",
	"pollthrottle" : 40,
	"queuethrottle" : 100,
	"sysqelemthrottle" : 1000,
	"refreshrate" : 33.333332061767578,
	"schedslop" : 25.0,
	"eventprobing" : 0,
	"mixerparallel" : "off",
	"mixercrossfade" : 0,
	"mixerlatency" : 30.0,
	"mixerramptime" : 10.0,
	"videoengine" : "viddll",
	"glengine" : "gl2",
	"packages" : 	{
		"BEAP" : "1.0.3",
		"dp.kinect2" : "1.2.20200629",
		"hap" : "1.0.6",
		"jit.mo" : "1.1.2",
		"max-mxj" : "7.2.5",
		"maxforlive-elements" : "0.3.7",
		"Mira" : "1.2.1",
		"Node For Max" : "1.3.3",
		"VIDDLL" : "1.2.4",
		"Video & Graphics" : "7.2.0",
		"Vizzie" : "2.1.4"
	}

}

## Steps to reproduce

Please detail every step needed to reproduce your issue.

*** I create a new patch
I instantiate a dp.kinect2 object
I right click and open dp.kinect2help 
I go to register tab
I enter my user name John Daniel
i click register
i get the above errors

Set `@verbose 1` on dp.kinect2 to receive additional debug information in the Max console.
I recommend you create a small patch that demonstrates the problem. Attach your patch to this issue.
Here is an example of detailed steps...

1. Start Max 8.
2. Open the dp.kinect2 help patch
3. Click on the xyz tab
4. Press blue button
5. Click the checkbox labeled "start"
6. ...

## Result
No DP kinect working

## Expected Result

DP kinect working

## Workarounds
NONE I could find
Did you discover any workarounds? What are they?

## Max console output

```
When you set `@verbose 1`, additional debug messages were sent to the Max console.
Please copy and paste here...
1. The first group of messages from dp.kinect2 which provide version information and settings.
2. Include any diagnostic messages you find related to your issue. They will be in yellow/red color.
```

diablodale, 2021-09-13:
Hello. I don't yet have enough information to help. You skipped answering several questions I request you revisit them.
I have not yet had a registration scenario that could not be resolved. So lets get started. 🙂

Please keep as reference the setup, hardware, and registration info at https://github.com/diablodale/dp.kinect2/wiki#hardware-requirements. 
### What I know now

* I suspect you are running an old version of dp.kinect2 as I see above `"dp.kinect2" : "1.2.20200629"`. Please update to the current version from https://hidale.com/shop/dp-kinect2/
* !! Ensure !! when typing your registration name, type it by hand. No copy/paste because sometimes extra characters are included. Ensure there is no extra whitespace, alternate versions of characters, etc. Using the correct and clean registration name is often a problem with registering. You received your registration name in one of the three emails my store sent you. Please use the registration name from that email.
* I tested myself your same download ZIP file with the current dp.kinect2 from https://hidale.com/shop/dp-kinect2/. I followed your exact repro steps. I had no errors and successfully registered. This suggests to me it is something local to your computer or local to your registration process. 
* I doubt your download ZIP file (and the download dpreg within it) are the problem. I see you downloaded your ZIP file twice. It is near impossible for your browser to twice fail, twice get ZIP files that pass ZIP's internal verification, yet the file within the ZIP be corrupted. Your dpreg file size is 805 bytes. If you also have 805 bytes, there is no realistic possibility it is corrupted. Please be sure you are first extracting files from the ZIP download, and then choosing the dpreg file just extracted.
* The error code you received suggest there is something wrong with the core Windows encrypt/decrypt APIs. I can not exactly pointpoint the place and specific issue. Only that the problem is occuring in some part of the process to encrypt and/or decrypt.
* I see you have not registered Max. I am not aware of ay dp.kinect2 issues when Max is run in demo, readonly, or trial modes. Enough years of success rule out almost anything related. Still, I recommend you register Max so that baseline can be established.

### Info needed

1. Is this same computer with the same setup that you successfully used for the dp.kinect2 14-day trial?
2. Did you make any computer configuration changes after the 14-day trial and before you attempted to register your paid license?
3. What is your dp.kinect2 version? This appears in the Max console. Please follow your repro steps and get the version that appears in the Max console. DO NOT provide the version from the mxe64 file or its ZIP name download. Get the version from the Max console. (Why? Because Max may be loading an old unexpected version of mxe/mxe64 files)
4. Please provide all the Max console output when you follow your repro steps. Please follow your repro steps, and then provide all of your Max console output -- top to bottom. You are welcome to redact sensitive 3rd party info. Still include all output from dp.kinect2.
5. I doubt it is the issue... yet... are you current with Windows Updates? Are you currently virus free?
6. The Windows OS version info you have provided is self-conflicting. In one place you write `build 19041` in another `build 19042`. Why is this?

jdjddjdj, 2021-09-13:
Thanks Dale !
I will go thru this line by line late tonight and get back with you.
Thanks so much for your help
Cheers
Jd

jdjddjdj, 2021-09-14:
Hi Dale - I only got the chance to look at this for a few minutes but found
1 definite problem

On downloading my registration key I must have right-clicked the link and
saved my dp.kinect2.dpreg link as an html file - Thusly never not getting
the proper zip file.  ( Doh !!!)

When i show file extensions the name shows as: "dp.kinect2.dpreg.htm" and
has a Firefox Icon.
File size is 2725 bytes and 4096 bytes taken on disk

Have to go to a job, but will be be back on much later with the rest of the
information you need.
thanks you for your help on this.
cheers
jd

jdjddjdj, 2021-09-15:
**### _VERBOSE Max Console Output
>New Patcher
>Instantiate dp.kinect2
>right Click > Open DP Kinect help
>Copy all in patch on first tab of help
>Close all maxpats
>New from clipboard
>added the attribute @verbose to the dp.kinect2 object
> patch saved as For_DP.maxpat ( attached to this message)
>Quit and Restart Max
>Open For_DP.maxpat

Console results:
dp.kinect2 1.2.20210521+723f5c9 Win64, Copyright (c) 2014-2021 Dale Phurrough. This software comes with ABSOLUTELY NO WARRANTY.
-All rights reserved except for those documented in the dp.kinect2 terms at https://hidale.com and granted by purchase/agreement to any entity listed as registered below.
-Compiled with Eigen 3.3.9 licensed under Mozilla Public License, v2.0. Source code & license at http://eigen.tuxfamily.org/
-BY USING THIS SOFTWARE, YOU ACCEPT THESE TERMS. IF YOU DO NOT ACCEPT THEM, DO NOT USE THIS SOFTWARE.
dp.kinect2 is not registered or has expired. Register your key or visit https://hidale.com for purchase and support.

In patcher >" Click to start"

Console Results:
dp.kinect2: dp.kinect2 is not registered or has expired. Register your key or visit https://hidale.com for purchase and support.
dp.kinect2: General access denied error. hr=0x80070005

_______
_______
_______
_______

**_Will answer your questions back to me inline  with leading *****_**

_Hello. I don't yet have enough information to help. You skipped answering several questions I request you revisit them.
I have not yet had a registration scenario that could not be resolved. So lets get started. 🙂

Please keep as reference the setup, hardware, and registration info at https://github.com/diablodale/dp.kinect2/wiki#hardware-requirements.
What I know now

    I suspect you are running an old version of dp.kinect2 as I see above "dp.kinect2" : "1.2.20200629". Please update to the current version from https://hidale.com/shop/dp-kinect2/
_****** Updated**_
    !! Ensure !! when typing your registration name, type it by hand. No copy/paste because sometimes extra characters are included. Ensure there is no extra whitespace, alternate versions of characters, etc. Using the correct and clean registration name is often a problem with registering. You received your registration name in one of the three emails my store sent you. Please use the registration name from that email.
_****** Typed by Hand**_ 

    I tested myself your same download ZIP file with the current dp.kinect2 from https://hidale.com/shop/dp-kinect2/. I followed your exact repro steps. I had no errors and successfully registered. This suggests to me it is something local to your computer or local to your registration process.
    I doubt your download ZIP file (and the download dpreg within it) are the problem. I see you downloaded your ZIP file twice. It is near impossible for your browser to twice fail, twice get ZIP files that pass ZIP's internal verification, yet the file within the ZIP be corrupted. Your dpreg file size is 805 bytes. If you also have 805 bytes, there is no realistic possibility it is corrupted. Please be sure you are first extracting files from the ZIP download, and then choosing the dpreg file just extracted.

_******* The DPREG file i see i named: "dp.kinect2.dpreg.htm" and has a Firefox Icon.
File size is 2725 bytes and 4096 bytes taken on disk   This file is also attached
I believe this is the main problem.
I have attemped to re-download thedpreg file, but I get the message "Sorry, you have reached your download limit for this file Go to shop"**_

    The error code you received suggest there is something wrong with the core Windows encrypt/decrypt APIs. I can not exactly pointpoint the place and specific issue. Only that the problem is occuring in some part of the process to encrypt and/or decrypt.
    I see you have not registered Max. I am not aware of ay dp.kinect2 issues when Max is run in demo, readonly, or trial modes. Enough years of success rule out almost anything related. Still, I recommend you register Max so that baseline can be established.

Info needed

    Is this same computer with the same setup that you successfully used for the dp.kinect2 14-day trial?
_*****  **Yes**_
    Did you make any computer configuration changes after the 14-day trial and before you attempted to register your paid license?
_******* No**_
    What is your dp.kinect2 version? This appears in the Max console. Please follow your repro steps and get the version that appears in the Max console. DO NOT provide the version from the mxe64 file or its ZIP name download. Get the version from the Max console. (Why? Because Max may be loading an old unexpected version of mxe/mxe64 files)
_**********  I updated to the latest available version
dp.kinect2 1.2.20210521+723f5c9 Win64, Copyright (c) 2014-2021 Dale Phurrough. This software comes with ABSOLUTELY NO WARRANTY**.**_

    Please provide all the Max console output when you follow your repro steps. Please follow your repro steps, and then provide all of your Max console output -- top to bottom. You are welcome to redact sensitive 3rd party info. Still include all output from dp.kinect2

_******* All console output:
dp.kinect2 1.2.20210521+723f5c9 Win64, Copyright (c) 2014-2021 Dale Phurrough. This software comes with ABSOLUTELY NO WARRANTY.
-All rights reserved except for those documented in the dp.kinect2 terms at https://hidale.com and granted by purchase/agreement to any entity listed as registered below.
[... cut ...]
whereas Max > About Max > Copy support information to clipboard results:

"version" : "Version 8.1.11 (17de540) (64-bit windows)",
	"platform" : "windows",
	"arch" : "x64",
	"osversion" : "Microsoft Windows 10 Professional (build 19041), 64-bit",
	"samplerate" : 44100,
	"iovs" : 512,
	"sigvs" : 64,
	"scheduler_in_audio_interrupt" : "off",**_

[dp.kinect2.dpreg.zip](https://github.com/diablodale/dp.kinect2/files/7166618/dp.kinect2.dpreg.zip)
[For_DP.zip](https://github.com/diablodale/dp.kinect2/files/7166619/For_DP.zip)_**

diablodale, 2021-09-15:
I removed the files you uploaded in above message to ensure your license file and/or private information was not disclosed. I did review the 1st ZIP file before doing that.

The file suggests that your process saved the raw html from a page of my store's website after previously downloading the actual ZIP file multiple times and therefore reaching a limit. This means that your process download 3+ times.

However/whatever downloaded (first two times) the dpreg license file...didn't do so. It instead created an htm file. This isn't file corruption. Instead, the app or method you used for download is itself the issue. Perhaps Sophos is interfering with the download during its middleman review. My store is not sending you htm files when you download. Your process incorrectly manifested them.

1. I recommend you again open the email my store sent you using a well known and tested email app/website.
2. Follow the instructions provided by your email app/website to open the email, and left-click on the download link in that email. You can click in another way if your email app/website or- accessibility needs it. However, such click needs to result in a standard HTTP GET to that download link's url. Your email app/website and/or Sophos should guide you through the process to download the needed ZIP file.
3. `zip` is the file extension. Not htm, html, or any other extension. If your process gives you something other than `zip`, then the process again is interfering. I can't help you further on how to download files from links using your email app and virus tool. Those vendors provide that support to you.

I have enabled a 3rd download for you. The download link in the email should again work.
If you created an account on my store, you can also visit https://hidale.com/my-account/ and download from there.

jdjddjdj, 2021-09-15:
Hello Dale.
I am an idiot -

i logged into my account,
clicked on the dpreg link, my download option appeared…

 because i wanted to do a screen capture (just in case there were any more
issues), i hit cancel

 Apparently the store read that as a download, and my download limit is
reached.

I feel like an idiot.  Could you enable one more time?

Sorry to be trouble. Thanks for your time.
Cheers
Jd

diablodale, 2021-09-15:
No worries, like lucky clover 🍀 
Try it now

jdjddjdj, 2021-09-16:
Hello Dale!
Giddy is the word.   Thanks SO much. I got it installed and is working *
perfectly *.
Was able to accomplish today most of the project I have coming due.

Wonderful plugin.
thanks again for your help!

cheers
jd

closed 2021-09-16 by diablodale