diablodale/dp.kinect2 #60: macbook installed Windows by Bootcamp-  after restored windows 10(bootcamp)  - lost access to the windows 10 I registered

HanXuJohn12345, 2022-09-03:
## Description

Hi Dale again，

I have one Macbook and I installed Windows by Bootcamp.

And then, I reinstalled Windows on the same MacBook, and want to register dp.kinet which result in "Your registration key is already registered on two different computers.

It is only one Mac, which installed Windows 10 by Bootcamp (I have successfully registered your dp.kinect and used it on it). Later I restored windows 10 again by using Bootcamp (now lost access to the windows 10 I registered)

## Setup

* Description of your computer hardware

Macbook Pro 15-inch, Mid 2015
Processor:2.5 GHz Quad-core intel Core i7
Memory:16GB 16000 MHz DDR3
Graphics: intel Iris Pro 1536MB

* Version of your Operating System and any patches/updates
Windows 10 Pro - installed by Mac OS bootcamp

* Version of Microsoft's Kinect v2 drivers (or SDK) you have installed
KinectSDK-v.2.0

* Version of Cycling74 Max you have installed and if it is registered
Version 8.3.1(x64 windows)

* Version of the dp.kinect2 external you are using
dp.kinect2-v1.2.202110521

## Steps to reproduce

Please detail every step needed to reproduce your issue.
Set `@verbose 1` on dp.kinect2 to receive additional debug information in the Max console.
I recommend you create a small patch that demonstrates the problem. Attach your patch to this issue.
Here is an example of detailed steps...

1. Start Max 8.
2. Open the dp.kinect2 help patch
3. Click on the xyz tab
4. Press blue button
5. Click the checkbox labeled "start"
6. ...

## Result
[image]

What result did you *actually* get? Please provide screenshots when helpful.

## Expected Result

What result did you *expect* to get?

## Workarounds

Did you discover any workarounds? What are they?

## Max console output
sorry I can not see xyz tab and blue button. Please inform me where they are?Thank you
[image]

```
When you set `@verbose 1`, additional debug messages were sent to the Max console.
Please copy and paste here...
1. The first group of messages from dp.kinect2 which provide version information and settings.
2. Include any diagnostic messages you find related to your issue. They will be in yellow/red color.
```

diablodale, 2022-09-04:
The form is an example
> Here is an example of detailed steps...Start Max 8...xyz..blue button

You need to write step 1, 2, 3, 4, 5, 6, 7, 8, 9, ...
Very detailed. Exactly what I need to do for every step.

Next, I also need.......

## Result

What result did you *actually* get? Please provide screenshots when helpful.

## Expected Result

What result did you *expect* to get?

## Workarounds

Did you discover any workarounds? What are they?

## Max console output

```
When you set `@verbose 1`, additional debug messages were sent to the Max console.
Please copy and paste here...
1. The first group of messages from dp.kinect2 which provide version information and settings.
2. Include any diagnostic messages you find related to your issue. They will be in yellow/red color.
```

HanXuJohn12345, 2022-09-04:
## Steps to reproduce

1. I was using macbook by bootcamp to install windows 10 pro on my mac
3. I was getting succeed about register dp.kinect on my Max/MSP
4. I forgot unregister on my max/msp, and re-installed new windows 10 pro on same macbook by bootcamp
5. when I use my new windows 10 pro on same macbook, My max/msp could not register dp.kinect because I could not get access my old windows 10 pro to unregister dp.kinect ( with max/msp on it)

## Result

can not register from my Max/MSP in my current windows 10 

## Expected Result

I expect that I can register on my new window 10 pro

## Workarounds

I do not know how to resolve the issue that I have

## Max console output

```
When you set `@verbose 1`, additional debug messages were sent to the Max console.
Please copy and paste here...
1. The first group of messages from dp.kinect2 which provide version information and settings.
2. Include any diagnostic messages you find related to your issue. They will be in yellow/red color.
```
[image]

diablodale, 2022-09-04:
I don't see any problems with dp.kinect2. It is currently unregistered and you have not shown any error messages of unsuccessful attempts at registration in your error messages or screenshots.

I do see a Max and Windows OS error: "Error 126 loading external jit.qt.engine". I have seen this before with Max. And others on the internet have seen this too https://www.google.com/search?q=Error+126+loading+external+jit.qt.engine I recommend you fix Max and Windows first.

Did you follow all setup steps? https://github.com/diablodale/dp.kinect2/wiki#setup-guide
I do not see anywhere the setup steps you followed. For example, you don't list you did setup step 3.

Last, you still have not provided me detailed steps to reproduce your problem.
Your steps are not detailed enough for me to understand what you are doing.
You need to write step 1, 2, 3, 4, 5, 6, 7, 8, 9, ...
Very detailed. Exactly what I need to do for every step.
EVERYTHING
what did you type?
what did you click?
what order did you click and type?
and so on...

Pretend I am blind and I am a child. Write EVERYTHING that I need to do to reproduce your problem.

HanXuJohn12345, 2022-09-05:
Hi Dale，

Thank you very much for your patient.

I really appreciate it.

I followed all the setup steps from this.  https://github.com/diablodale/dp.kinect2/wiki#setup-guide

I know I have "Error 126 loading external jit.qt.engine" but this is not the main problem.

The problem is that I could not register on Max/MSP from my new windowsOS now 
because I did not unregister on my Max/MSP from uninstalled windows OS
*It should be notice that I only have one MacBook， my new windowsOS  is on this Macbook, and my uninstalled windows OS was on same the same MacBook.

Is this clear for my main problem's description?

I will list set up my process below

0. I was using MacBook by Bootcamp to install windows 10 pro on my max/MSP
1.（did) Install [Cycling74 Max 6.x (32 or 64-bit)](http://cycling74.com/downloads/) or newer.
2.（did) Install [Kinect v2.x Runtime](https://www.microsoft.com/en-us/download/details.aspx?id=44559).
3.（did) Install the most recent Microsoft Visual C++ 2015 Redistributable from [secondary](https://www.microsoft.com/en-us/download/details.aspx?id=53587) links. Be sure to install the 32 or 64-bit version (or both) to match your Cycling74 Max platform version.
4. （did) Plug your Kinect for Windows v2 sensor into a USB 3.0 port. Please do not plug any other USB devices into the same USB 3.0 controller because the Microsoft Kinect v2 drivers need all the bandwidth.
5. （did) Download dp.kinect2 from http://hidale.com/shop/dp-kinect2/#download.
6. （did) Open the ZIP archive and move the whole dp.kinect2 folder into your packages folder. The default Max packages folders are
7.（did) Max 8: home directory -> Max/Max8 -> Packages folder

For register part

1. And Then when I restart Max/MSP, I was able to use dp.kinect object.4. 
2. I got your email to download xxxxx.dpreg file.
3.  I was using dp.kinect help file, enter the user name that you send  to me to register successfully(click register with my user name, select your xxxxx.dpreg file, and register successfully)
4. I was getting successful about register dp.kinect on my Max/MSP5. 
5. Next day, I forgot to unregister on my max/MSP, and re-installed the new windows 10 pro on the same MacBook by Bootcamp.
6. Then, when I use my new windows 10 pro on the same MacBook, My max/MSP could not register dp.kinect because I could not get access to my old windows 10 pro to unregister dp.kinect ( with max/MSP on it)

## Result

I can not register from my Max/MSP in my current windows 10

## Expected Result

I expect that I can register on my new Windows 10

## Workarounds

I really do not know how to deal with my problem

## Max console output
[image]

```
When you set `@verbose 1`, additional debug messages were sent to the Max console.
Please copy and paste here...
1. The first group of messages from dp.kinect2 which provide version information and settings.
2. Include any diagnostic messages you find related to your issue. They will be in yellow/red color.
```

diablodale, 2022-09-05:
Thank you, I still need more information.

I need ALL the steps for "My max/MSP could not register dp.kinect".
All the steps. For example...this is only an example
1. I ran Max
2. I opened the dp.kinect2.maxhelp file
3. I clicked on the tab xyz123
4. I clicked on the button xyz123
5. and so on...

Your screenshot is not enough. You are hiding important information.
Please copy and paste here...
1. The first group of messages from dp.kinect2 which provide version information and settings.
2. Include any diagnostic messages you find related to your issue. They will be in yellow/red color.

I recommend you switch to your Max console, press CTRL-A, CTRL-C, and then paste all your clipboard into a post here.

I can then match your detailed steps and the max console output.

HanXuJohn12345, 2022-09-05:
Hi Dale, I think that I understand what you mean.
this reproduction process is very detailed.

0. I was using MacBook by Bootcamp to install windows 10 pro on my max/MSP
1. Install [Cycling74 Max8.3.1(x64 windows)]
2. Install KinectSDK-v.2.0
3. Install the most recent Microsoft Visual C++ 2015 Redistributable from [secondary]  64-bit version 
4. Plug your Kinect for Windows v2 sensor into a USB 3.0 port. Please do not plug any other USB devices into the same USB 3.0 controller because the Microsoft Kinect v2 drivers need all the bandwidth.
5. Download dp.kinect2 from http://hidale.com/shop/dp-kinect2/#download.
6. Open the ZIP archive and move the whole dp.kinect2 folder into your packages folder. 
7. Max 8: home directory -> Max/Max8 -> Packages folder
8. I ran Max
9. I opened the dp.kinect2.max help file
10. And Then when I restart Max/MSP, I was able to use dp.kinect object.
11. I got your email to download xxxxx.dpreg file.
12. I was using dp.kinect help file, enter the user name that you send to me to register successfully(click register with my user name, select your xxxxx.dpreg file, and register successfully)
13. I was getting successful about register dp.kinect on my Max/MSP
14. Next day, I forgot to unregister on my max/MSP, and re-installed the new windows 10 pro on the same MacBook by Bootcamp.
15. Next day, I forgot to unregister on my max/MSP， and re-installed the new windows 10 pro on the same MacBook by Bootcamp.
16. Then, when I use my new windows 10 pro on the same MacBook.
17.  Install [Cycling74 Max8.3.1(x64 windows)] again
18. Install KinectSDK-v.2.0 again
19.  Install the most recent Microsoft Visual C++ 2015 Redistributable from [secondary]  64-bit version again
20. Plug your Kinect for Windows v2 sensor into a USB 3.0 port. Please do not plug any other USB devices into the same USB 3.0 controller because the Microsoft Kinect v2 drivers need all the bandwidth again
21. Download dp.kinect2 from http://hidale.com/shop/dp-kinect2/#download again
22. Open the ZIP archive and move the whole dp.kinect2 folder into your packages folder again
23. Max 8: home directory -> Max/Max8 -> Packages folder again
24. I ran Max again
25. I opened the dp.kinect2.max help file again
26. And Then when I restart Max/MSP, I was able to use dp.kinect object.again
27. I got your email to download xxxxx.dpreg file. again
28.  I was using dp.kinect help file, enter the user name that you send to me to register unsuccessfully(click register @verbose 1, with my user name, select your xxxxx.dpreg file, and register unsuccessfully)
[image]
[image]

diablodale, 2022-09-06:
Please copy and paste here...

The first group of messages from dp.kinect2 which provide version information and settings.
Include any diagnostic messages you find related to your issue. They will be in yellow/red color.
I recommend you switch to your Max console, press CTRL-A, CTRL-C, and then paste all your clipboard into a post here.

I can then match your detailed steps and the max console output.

In all your screenshots, you hide this information. I need all the max console output. For example, I need everything in this max console. Everything in green.
[image]

HanXuJohn12345, 2022-09-07:
dp.kinect2 1.2.20210521+723f5c9 Win64, Copyright (c) 2014-2021 Dale Phurrough. This software comes with ABSOLUTELY NO WARRANTY.
-All rights reserved except for those documented in the dp.kinect2 terms at https://hidale.com and granted by purchase/agreement to any entity listed as registered below.
-Compiled with Eigen 3.3.9 licensed under Mozilla Public License, v2.0. Source code & license at http://eigen.tuxfamily.org/
-BY USING THIS SOFTWARE, YOU ACCEPT THESE TERMS. IF YOU DO NOT ACCEPT THEM, DO NOT USE THIS SOFTWARE.
dp.kinect2 is not registered or has expired. Register your key or visit https://hidale.com for purchase and support.
kinect performance caution: power plan should be a high performance plan
kinect performance caution: power setting (USB selective suspend) should be disabled
kinect performance caution: power setting (Link State Power Management) should be off
kinect performance caution: power setting (Minimum processor state) should be 100%
Error 126 loading external jit.qt.engine
dp.kinect2: dp.kinect2 requires Kinect20.Face.x64.dll 2.0.1410.19000 or greater + the matching NuiDatabase folder from that release
Gen working in runtime mode
dp.kinect2: dp.kinect2 is not registered or has expired. Register your key or visit https://hidale.com for purchase and support.
dp.kinect2: General access denied error. hr=0x80070005
dp.kinect2: RG: 33558528 RV: 0
dp.kinect2: dp.kinect2 is not registered or has expired. Register your key or visit https://hidale.com for purchase and support.
dp.kinect2: RG: 33558528 RV: 0
dp.kinect2: dp.kinect2 is not registered or has expired. Register your key or visit https://hidale.com for purchase and support.
[image]

diablodale, 2022-09-07:
Hi. I continue to see the Max error `Error 126 loading external jit.qt.engine`. You will need to solve this first as this is a failure of Max itself and nothing to do with dp.kinect2.

I also see no error message "Your registration key is already registered on two different computers" in your screenshot or in the copy/paste you provided.

I only see the error message and code 33558528 that indicate you have not registered dp.kinect2 on this computer. Please follow the instructions at https://github.com/diablodale/dp.kinect2/wiki#register-dpkinect2 to register on this computer.

HanXuJohn12345, 2022-09-07:
Yes, I know, The main issue is that I can not unregister on the last windows OS because I uninstalled it. So I am not able to not go to the max/MSP on the previous windows OS because I uninstalled this window from my computer.  no access anymore 

Could you please help me unregister it? In this way, I will be able to register on my new windows.

HanXuJohn12345, 2022-09-07:
Yes, I know, The main issue is that I can not unregister on the last windows OS because I uninstalled it. So I am not able to go to the max/MSP on the previous windows OS because I uninstalled this window from my computer. no access anymore

Could you please help me unregister it? In this way, I will be able to register on my new windows.

diablodale, 2022-09-07:
Please follow the instructions at https://github.com/diablodale/dp.kinect2/wiki#register-dpkinect2 to register on this computer.
The registration system can manage this automatically for you as described at https://github.com/diablodale/dp.kinect2/issues/27#issuecomment-243550690

HanXuJohn12345, 2022-09-07:
**If you did major erasing on those computers (e.g. formatted your drive, reinstalled Windows, deleted files and the registry, etc.), you might have erased the locally-stored registration information.

Easy fix :-) You can...

Register on those two computers using the "register" tab in the help file. This register will authorize with the licensing server and the server will see it has already authorized that computer. It will reuse and re-download the existing registration information. This is similar to a "resync".
Unregister using the "register" tab in the help file. The computer has its locally-stored registration information from step 1. It should be able to unregister.
With the above two steps, you will have unregistered your license and now you can move it to a new computer.

If you have problems in the above two steps, please provide similar diagnostic information that you provided above so that I can troubleshoot.**

I can not unregister anywhere
I can not register here as well

here I clicked unregister
And I clicked register with your .dpreg file

console shows these information

[image]

diablodale, 2022-09-07:
We may be having a language barrier, so I am repeating for the 3rd time...
Please follow the instructions at https://github.com/diablodale/dp.kinect2/wiki#register-dpkinect2 to register on this computer.

if you continue to have problems after you have followed the instructions at https://github.com/diablodale/dp.kinect2/wiki#register-dpkinect2 to register on this computer, then please provide the information that I continue to ask from you
1. exact steps
2. exact max console output

The screenshots of max console output do not match what you write. Therefore, you are not following the exact registration steps when you make that screenshot.

diablodale, 2022-09-07:
Also, I notice in your screenshots that you are not correctly doing steps 3 and 5.
https://github.com/diablodale/dp.kinect2/wiki#register-dpkinect2

Please follow all steps at https://github.com/diablodale/dp.kinect2/wiki#register-dpkinect2 and it is important that you correctly follow steps 3 and 5.

HanXuJohn12345, 2022-09-07:
Oh, Thank you so much for this. I got registration You are very kind. Thank you so much

diablodale, 2022-09-08:
Glad it is working for you.
Thank you for choosing dp.kinect2 for your solutions :-)