diablodale/dp.kinect2 #61: Depthmap with @type long gives unexpected result

Teufeuleu, 2023-01-12:
## Description

I use [dp.kinect2 @depthmap 1 @type long] in order to perform bitwise operations on the depth matrix. The main goal is to record a video I can use as in the [three.js kinect webgl example](https://threejs.org/examples/webgl_video_kinect.html), which implies encoding depth as RGB values (as shown in [this modified freenect file](https://gist.github.com/mrdoob/1326080#file-gistfile1-c-L295)).

However, the long matrix output seems to only be the truncated values of the float32/64 output. As a proof, a [jit.3m] gives, say 0 as min, 1.9 as mean and 7.9 as max for a float32/64 output, while the long output gives 0, 1.4 and 7 respectively.
I would expect the long output to keep the same resolution as a float32/64 output, and not being a truncated version of it with the equivalent of a 3bit resolution.

## Setup

* Windows 10 (with last service pack and updates) laptop, i7 8750H, 16GB RAM, GTX1060
* Version of Microsoft's Kinect v2 drivers (or SDK) you have installed : I have v2.0_1409 SDK and both v2.0_1409 and v2.2_1905 runtimes installed, not sure which one is relevant/being used by dp.kinect2
* Max 8.5.2
* dp.kinect2 1.2.20210521+723f5c9

## Steps to reproduce

1. Start Max 8.
2. Open the dp.kinect2 help patch
3. Set @depthmap 1 @type long as attributes of [dp.kinect2]
4. Create a [jit.3m] or [jit.cellblock] and connect its input to the leftmost output of [dp/kinect2]
5. Press the trigger to open the kinect stream
6. Check the outputs given by [jit.3m] or [jit.cellblock], and compare those to when @type is set to float32/64.

## Result

The long matrix output seems to only be the truncated values of the float32/64 output.

## Expected Result

I would expect the long output to keep the same resolution as a float32/64 output, and not being a truncated version of it with the equivalent of a 3bit resolution..

## Workarounds

A workaround might be to somehow translate the float64 output as a long, but I'm not sure how to do it without loss of information. All my attempts have failed so far.

## Max console output with @verbose 1:

Nothing special (last 3 lines are only when opening the device, then it runs smoothly)

```
dp.kinect2 1.2.20210521+723f5c9 Win64, Copyright (c) 2014-2021 Dale Phurrough. This software comes with ABSOLUTELY NO WARRANTY.
-All rights reserved except for those documented in the dp.kinect2 terms at https://hidale.com and granted by purchase/agreement to any entity listed as registered below.
-Compiled with Eigen 3.3.9 licensed under Mozilla Public License, v2.0. Source code & license at http://eigen.tuxfamily.org/
-BY USING THIS SOFTWARE, YOU ACCEPT THESE TERMS. IF YOU DO NOT ACCEPT THEM, DO NOT USE THIS SOFTWARE.
dp.kinect2 registered for [name]. Valid 12 Jan 2023 - 31 Dec 2999. Sn 1680.
kinect performance caution: power setting (USB selective suspend) should be disabled
dp.kinect2: data unavailable for request
dp.kinect2: data unavailable for request
dp.kinect2: data unavailable for request
```

diablodale, 2023-01-12:
That is (technically) expected behavior. I recommend you use `@distmeter 0` so that you receive millimeters with your `@type long`

Depth `long` output is an integer and has no fractional values. Therefore its possible values when `@distmeter 1` are only 0,1,2,3,4,5,6,7,8. With `@distmeter 1` output is in millimeters and therefore can be the integer range 0-8000.

Teufeuleu, 2023-01-13:
I knew I missed some attribute. Thanks!

closed 2023-01-13 by Teufeuleu

diablodale, 2023-01-13:
No worries. I've added this scenario to the new documentation (work in progress)
https://docs.hidale.com/max/attributes/distmeter/