diablodale/dp.kinect2 #62: dp.kinect does not work in ableton. The M4L window closes as soon as this object is inserted

MartinDDrum, 2023-10-11:
## Description
I want to use dp.kinect in Ableton Live and every time I try- it crashes. I was told that all dp objects are made to work in Ableton; however, I tried inserting this object in any of the three M4L devices and none of them wanted to accept it.

it works perfectly fine in Max!
thanks for your help making this work!

## Setup
* i7, RTX 2060, 64GB ram
* XPS 17, Windows 11 Home 22H2
* Max MSP 8.3.1
* dp.kinect2 1.2.202110521+723f5c9

## Reproduce steps
1 open Ableton
2 create new M4L device
3 insert the dp.kinect object

## Result
the M4L window freezes and it closes shortly after.

## Expected
I am expecting it to be able to run inside of Ableton. 

## Workarounds
the only workaround I can think of is to reroute midi out from max to Ableton with LoopMIDI

diablodale, 2023-10-11:
Hi. Please clarify several things...😊

1. You have opened this issue in the `dp.kinect2` area. Yet in your issue you write both `dp.kinect` and `dp.kinect2`. Please ensure the issue is in the correct area...or adjust your text to match.
2. You have not added all setup and repro information. List everything. Every file copied, installed, unzipped, where they go, how they are named, etc. I need to be able to follow exact steps that you list on my computer. What you don't list...I don't do and therefore can not help you.
3. You wrote in email "crash". But in your issue above you don't write that. Do you get any error messages or crash dialogs? If you do, please make a cropped screenshot of that dialog or error. Or if you can get the exact text of the dialog/error, that works also.
4. What version of Windows? Please type `ver` at a command prompt to get the full verson.
5. What version of Ableton?
6. The version of dp.kinect2 you provided does not exist. Did you add an extra number?
7. What is registered? Windows, max, ableton, max for live, etc.
8. Your repro step1 is ok. thank you.
9. Your repro step 2, please add detailed steps. I click on what menu or button, I do what next, and then what next...
10. Your repro step 3, how do I insert? Do I click a menu, right-click something, mouse click somewhere, etc.

MartinDDrum, 2023-10-11:
> Hi. Please clarify several things...😊
>
> 1. You have opened this issue in the `dp.kinect2` area. Yet in your issue you write both `dp.kinect` and `dp.kinect2`. Please ensure the issue is in the [...]

Thanks, I reported more information!

diablodale, 2023-10-11:
Please add specific detail for how you setup dp.kinect2. Your text `The dp files are in my external folder in Max 8 in documents.` doesn't match any known folders. Please list the multiple individual detail and sequential steps on how you installed dp.kinect2. I suspect you: 1) download from hidale.com, 2) unzipped... and so on.

I watched the video andI don't see a crash. So I'll not use the description you provided in email and only use what you provide here in this issue.

MartinDDrum, 2023-10-11:
I edited the text to be extremely specific, 
please use any of this information to help me through this!

"In computing, a crash, or system crash, occurs when a computer program such as a software application or an operating system stops functioning properly and exits." perhaps there is a better terminology for what is happening here?

diablodale, 2023-10-13:
Please install dp.kinect2 by the instructions at https://docs.hidale.com/max/dp-kinect2#setup
You did not follow..
* step 3. `Externals` is not a recognized folder by Max 8 for packages. Please follow the instructions for step 3.
* step 4 to install the runtime (which do or did you install?)
* step 5 to install the vs 2015 (which do or did you install?)

diablodale, 2023-10-23:
Closing issue as the customer has not responded.

closed 2023-10-23 by diablodale