diablodale/dp.kinect2 #64: Unable to unregister dp.kinect2 license

pantracols, 2024-09-12:
## Description

I installed my dp.kinect2 license in a computer that I wanted to use for a specific project, then I moved the license to a different windows user in the same computer, unregistering from the first one and then registering on the other one. I'm not sure what has happened but now it seems like the license is not registered on either user, but when I try to register it in either of the users or in a different computer, MAX says that the license has too many activations.

I haven't installed it anywhere else, I don't know where it could be activated. Is it possible for you to remove any activations from the license so I can register it on a new computer?

## Setup

* Computer hardware description: CPU, memory, storage, GPU, etc.
Asus Vivobook laptop 
12th Gen intel(R) Core(TM) i7-1255U 1.70 GHz
RAM: 16GB

* Operating system version. Run `winver` and provide the complete "Version xxxx (build xxxxxx)".
Windows 11 Version 23H2 (build 22631.2880)

* Cycling '74 Max version. Is Max registered?
Max 7. Version 7.3.5 (be88c5a) (64-bit windows)

* dp.kinect2 plugin version. Get it from the Max console when you load the plugin.
dp.kinect2 1.2.20210521+723f5c9 Win64

* ...and any remaining setup/configuration.

## Reproduce steps
any MAX patch with a dp.kinect2 object will generate a message in the Max console saying that it is not registered or has expired

I try to register:
open help
go to register tab
type my username
click on register
choose the file dp.kinect2-689.dpreg

## Max console output
I get the following message in the MAX console:
dp.kinect2: Unable to create activation request: registration key has too many activations; need to unregister on other computers 2024-09-12T18:40:07Z
dp.kinect2: RG: 167772160 RV: 0

setting `@verbose 1` doesn't provide any additional information in the MAX console

diablodale, 2024-09-13:
Hi. I looked and I see your license is registered already twice. They were each registered on...

* Monday, July 18, 2016 15:23:35  UTC
* Tuesday, June 25, 2019 11:49:03  UTC

I can see it is two different computers. However, I can not identify or describe the computers -- that information is one-way hashed to protect my customer's privacy.

You will need to unregister dp.kinect2 on these two computers. Then you can register your license on one new computer.

* https://github.com/diablodale/dp.kinect2/issues/9 is the usual method to move your license
* https://github.com/diablodale/dp.kinect2/issues/27#issuecomment-243550690 is a method when you have reformatted, cleaned, changed users, etc.

pantracols, 2024-09-13:
Hi. Thank you for your reply
I don't have access to those computers. They belonged to an institution I used to work for. The computers have been destroyed.
Is there a way to unregister without having access to the computers?
Thanks
Best
S

diablodale, 2024-09-13:
I treat everyone equally and fair. My customers are responsible for managing their own licenses. You are describing a scenario in which your license was over-registered on two computers. And then both computers destroyed with the license still in them, instead of removing it before destruction.

If I were to remove your two registrations, then I am providing you two free licenses so that you can register again on a third computer. In addition, I have no method to verify the first two computers are destroyed. They could be working computers in a long-running role or in a different department of that institution...still with dp.kinect2 being used on them.

This is not something that I am willing to make an exception. It would not be equal and fair to other customers for you to get three registrations when you only paid for one.

pantracols, 2024-09-13:
I understand what you say. However I'm not asking to have three active registrations, only one. You have no way to verify that the first two computers are destroyed. But you can verify that the license is mine, and so I can decide where to have it active regardless of whether they are being used by the institution or not. 

All software manufacturers let you retrieve licenses from computers you don't have access to anymore, as long as you can prove that the license is yours. Otherwise, if a computer dies, gets stolen or whatever, you'd just loose the license you payed for.

diablodale, 2024-09-13:
There is no provision in the licensing of dp.kinect2 for you to arbitrarily and remotely make a license active or deactive. Instead, the terms https://hidale.com/terms/ writes "The licensing cost of my software is related to the number of computers that have it installed".

I believe you own a license for installation and use of dp.kinect2 on one computer. Currently, your license is installed on two computers. You already exceeded the limit of one in June 2019. Your computer did not "die, gets stolen or whatever". You admitted that you over-registered and then did not remove the over-registered license that you own before you lost control of those computers. This is a problem of your own creation....not one that I am responsible to fix by providing unfair treatment.

pantracols, 2024-09-13:
I exceeded the limit in 2019 unknowingly, I just changed computers and I guess I didn't unregister the license correctly before installing in the new computer.

Yes, I created the problem, I just don't understand why you don't want to fix it, since you said that you can and you know that I own the license. 

So I should just buy a new license and loose the other one?

diablodale, 2024-09-13:
My answers remain the same.

> This is not something that I am willing to make an exception. It would not be equal and fair to other customers for you to get three registrations when you only paid for one.

At any time of your choice, you can visit those two computers. Unregister them both using the methods I provided. Then, you can use your license on any one computer that you want.

I hear that you are not pleased with my answers. However, it is important for me to respect all my customers equally and fairly by consistently applying the normal and stable terms https://hidale.com/terms/

> I do allow transfer of licenses from one computer to another computer under your (the purchaser’s) control. It is an honor system; I am trusting you. For example, after your development or performances, you could unregister the software on any computer and then use those licenses on any other computers you control. For example, you might use this method to move your license with dp.kinect2.

pantracols, 2024-09-13:
I can't visit those computers because they don't exist anymore, they were thrown away, the institution is not an art center anymore and so I was fired. 

This has nothing to do with fairness, it just doesn't make any sense.

diablodale, 2024-09-14:
dp.kinect2 is working correctly, and the license is correctly enforcing the terms of use. There is no issue for which I am responsible to resolve. Please open a new issue if you have problems with setup, registration, usage, or technical problems with dp.kinect2.

I will close and remove this issue later this week since the topics discussed are external to dp.kinect2.